nasa/cFE #824: Banned functions are used, snprintf

ArielSAdamsNASA, 2020-08-20:
**Describe the bug**
A banned function, snprintf, is used. Banned due to potential overflow issues. 

**Expected behavior**
Use safer function such as sprintf_s. 

**Code snips**
https://github.com/nasa/cFE/blob/a148b97511de3f45f6677c1a3a0b17197f867e28/fsw/cfe-core/src/es/cfe_es_task.c#L381-L399

**Additional context**
References:
- https://docs.microsoft.com/en-us/previous-versions/bb288454(v=msdn.10)?redirectedfrom=MSDN
- https://docs.roguewave.com/en/klocwork/current/sv.banned.required.sprintf
- https://cwe.mitre.org/data/definitions/676.html
- https://help.semmle.com/wiki/display/CCPPOBJ/Potentially+overflowing+call+to+snprintf
- http://tldp.org/HOWTO/Secure-Programs-HOWTO/dangers-c.html

**Reporter Info**
Ariel Adams, ASRC Federal

jphickey, 2020-08-20:
Is it `snprintf()` that is actually banned?  `sprintf()`, for sure ... and most the references above are talking about `sprintf` and suggesting use of `snprintf()` as the safer alternative.

ArielSAdamsNASA, 2020-08-20:
@jphickey You are right that `snprintf()` is a much better alternative than `sprintf()`, however, there are still issues with `snprintf()` that can be replaced with safer alternatives. The function does not guarantee protection against buffer overflows since it depends on the buffer size you provide. You can find the list of banned functions here: https://docs.microsoft.com/en-us/previous-versions/bb288454(v=msdn.10)?redirectedfrom=MSDN. Here is an additional link for more information: https://docs.microsoft.com/en-us/cpp/c-runtime-library/reference/snprintf-snprintf-snprintf-l-snwprintf-snwprintf-l?view=vs-2019

[image]

jphickey, 2020-08-20:
@ArielSAdamsNASA  -- I'm still unclear on this -- the document specifically cited in your previous comment says that `sprintf()` is the banned function and `snprintf()` is the recommended alternative, unless I'm reading this incorrectly?  I can't find anything that says __snprintf__ is "banned" 

There have been cases of some library vendors not implementing it properly according to the standard.
And there is the issue of not handling the return code properly.

But when properly used I can't see how `snprintf` is inadequate/inappropriate for what we are doing here.

closed 2020-08-20 by ArielSAdamsNASA

astrogeco, 2020-08-20:
@ArielSAdamsNASA do we need to close this issue or just figure out how to make sure that our use of `snprintf` don't result in buffer overruns?

thnkslprpt, 2022-10-17:
> I concur that `snprintf` has some caveats/dangers - particularly with respect to the return code and handling that properly - whereas `sprintf()` is a buffer overflow exploit in almost all use-cases, and should rightfully be banned.

There are still a few stray uses of sprintf() across cFS. Should these be removed?