nasa/cFE #845: Unsafe macro in cfe_version.h

skliper, 2020-08-26:
**Is your feature request related to a problem? Please describe.**
Unsafe macro (violates coding standard)

https://github.com/nasa/cFE/blob/06c08268f2d7a7fe72aceec8c33201ce6ccba2f4/fsw/cfe-core/src/inc/cfe_version.h#L48

**Describe the solution you'd like**
Add parentheses

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-08-26:
Nevermind.  Read up on it, since it quotes it's safe.

closed 2020-08-26 by skliper

astrogeco, 2020-08-26:
> Nevermind. Read up on it, since it quotes it's safe.

Can you add the stuff you read so we can reference it if it comes up again? I was actually discussing this earlier and wondering if we need to have the integer macros at all. If they're all strings then we wouldn't need the helper functions.

skliper, 2020-08-26:
My preference is to remove any unused macros!  I just brushed up on the preprocessor macro at https://www.cprogramming.com/reference/preprocessor/stringizing-operator.html, and wrote a quick test program to try and break it and I couldn't (just to confirm it behaved as expected on random strings).