nasa/cFE #854: Cast align issue in cfe_es_task.c - CFE_ES_TaskPipe

skliper, 2020-08-31:
**Describe the bug**
```
/home/jhageman/cFS-GitHub/cfe/fsw/cfe-core/src/es/cfe_es_task.c: In function ‘CFE_ES_TaskPipe’:
/home/jhageman/cFS-GitHub/cfe/fsw/cfe-core/src/es/cfe_es_task.c:635:52: warning: cast increases required alignment of target type [-Wcast-align]
  635 |                         CFE_ES_SendMemPoolStatsCmd((CFE_ES_SendMemPoolStats_t*)Msg);
```

**To Reproduce**
Build on gcc 8+ with cast-align=strict

**Expected behavior**
Clean build

**Code snips**
https://github.com/nasa/cFE/blob/06c08268f2d7a7fe72aceec8c33201ce6ccba2f4/fsw/cfe-core/src/es/cfe_es_task.c#L632-L637

**System observed on:**
 - Hardware: arm64
 - OS: Ubuntu 20.04
 - Versions bundle + mods to cast-align=strict

**Additional context**
#851, #852 

**Reporter Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-08-31:
@jphickey are you fixing this with the other mempool command issues (reference instead of raw address)?

jphickey, 2020-08-31:
Yes, the reason this message has increased alignment needs is because it contains a `cpuaddr` type.  When replaced with an abstract pool handle (which can be `uint32`) the alignment issue goes away.

skliper, 2021-01-06:
Confirmed this has been resolved (via #917).  Closing

closed 2021-01-06 by skliper