nasa/cFE #855: Have "make test" provide full stats on the pass/fail ratio of the asserts

asgibson, 2020-09-01:
**Is your feature request related to a problem? Please describe.**
When running ```make test``` it only gives information about which testrunner had at least 1 failing assert.  This is useful, but very limited.

**Describe the solution you'd like**
Each test file run by ```make test``` provide a "fail asserts / total asserts" in the output.

**Describe alternatives you've considered**
Running each testrunner individually to get these results.
Writing my own bash script.

**Additional context**
Could be added as an additional runner ```make test/stats```?

**Requester Info**
Alan Gibson NASA GSFC/587

skliper, 2020-09-01:
Have you considered using the test log(s)?  It's got all the information you are requesting from a `make test`.

If that doesn't work, can you quantify the value added by this requested change?  I haven't experienced a case where I would have done anything differently based on getting more information from `make test`.  If there's a failure, I examine the log and/or rerun the test.  I mostly use `make test` from a CI (pass/fail) point of view vs a measure of how many failures (or asserts) occur.

skliper, 2020-09-01:
By log(s) I mean build/native/default_cpu1/Testing/Temporary/LastTest.log and LastTestError.log (I think that's what it's called, only shows up on errors).  Or whatever structure your build takes depending on config.

jphickey, 2020-09-01:
Note that `make test` in turn just runs `ctest` which is a third party program, part of the CMake suite.  This is in turn actually running the tests and reporting pass/fail.  We don't control the code for this, so the only means we have to make it output something different is via its command line options.

In general the only acceptable failure rate is zero, so hence why something like "make test" with only a simple pass/fail is good enough most of the time, just to tell you whether something needs attention or not.

Also there should be a log file (`-O` command line option to ctest) and as such one should be able to "grep" that file after the fact and get the specific numbers you are looking for.

asgibson, 2020-09-02:
> Have you considered using the test log(s)? It's got all the information you are requesting from a `make test`.
>
> If that doesn't work, can you quantify the value added by this requested change? I [...]

When developing this can be a quick resource to see how much damage has been done by my changes, i.e. if 1 out of 100 tests are failing, then the damage is minimal vs now ALL 100 tests are failing.  I admit that maybe this is only informational, but it can speed up troubleshooting.  Unit test frameworks I have used often provide this output no matter the size of a test run.

> By log(s) I mean build/native/default_cpu1/Testing/Temporary/LastTest.log and LastTestError.log (I think that's what it's called, only shows up on errors). Or whatever structure your build takes depending on config.

The LastTestError.log only lists what already is output by ctest3, the names of testrunners that have failures.  The LastTest.log does have run information in there (just like running a single test output), but it is only "the last test run" which may or may not have been a failed testrunner (in my case the utils tests because of 'u').  Not very helpful.

> Note that `make test` in turn just runs `ctest` which is a third party program, part of the CMake suite. This is in turn actually running the tests and reporting pass/fail. We don't control the code [...]

While I agree that the only acceptable case is 0, that is when things are complete, not during development of code or unit tests.  When I am developing these, this information available to me in the "run all tests" format is helpful.  When I use test driven design strategies, I will run this over and over locally as I develop to ensure I am not breaking things.  When I do, which is inevitable, knowing how bad I messed things up may point out my mistake. 

I can figure out a local solution (I will look into the -O option which is in the sample Makefile `make test` but not in the unit-test dir's automatically built Makefile) and will work on it "as I can while still getting my work done."  I may even be able to provide the overall solution later; however, I thought maybe it already existed or someone already had an idea of how to do this so I put in this feature request to see if that were the case.  If it will only be valuable to me then disregard.

skliper, 2020-09-02:
You run `make test` when doing test development?  I find it much more efficient to just run the test I'm interested in.  That would save you minutes every cycle, and contains all the information you are requesting...

skliper, 2020-09-02:
Also note, I checked LastTest.log and it contains all 76 tests that were run (~11k lines)... so if you only see the last test case there's something that isn't working as expected.

asgibson, 2020-09-02:
> You run `make test` when doing test development? I find it much more efficient to just run the test I'm interested in. That would save you minutes every cycle, and contains all the information you are requesting...

I am running it in the app's unit-test, it runs all the tests in the directory for my app.  I want to run this often, it is quick, painless and informative during development.  However, yes, I do.  Not often, given the length of time it takes (extremely painful), but I have to ensure things are in working order, trust but verify.

> Also note, I checked LastTest.log and it contains all 76 tests that were run (~11k lines)... so if you only see the last test case there's something that isn't working as expected.

Indeed, not sure what I was seeing before (had I only run 1 test last time, didn't think so, but clearly must have been the case), they do all all appear there.  Much appreciated on making me take another look.  Does that change my view on having this information automatically output, no.  Digging through my local file of 11 tests sill benefits from automation of output.

asgibson, 2020-09-02:
>You run make test when doing test development?

Also, it's the easiest way to check my current coverage percentage?  The Makefile in the built unit-test directory directory does not appear to provide coverage building.  I can probably do the lcov manually, I just hadn't considered that yet.

skliper, 2020-09-02:
Agreed that coverage on an individual run is very useful, which as you mentioned can be achieved w/ lcov and genhtml.
```
lcov -c --rc lcov_branch_coverage=1 -d ../.. -o coverage.info
genhtml --branch-coverage coverage.info
```

jphickey, 2020-09-02:
Yep - you can get coverage report for just one app by running `lcov` with an appropriate `--directory` option.

Note that the "top level makefile" which has the `make test` and `make lcov` target that invokes lcov is merely a convenience tool.  Users are more than welcome to add additional targets to this makefile for things that make their workflow more efficient.  This is precisely why it is distributed as `Makefile.sample` and not directly version controlled (you are forced to clone and own it, if you even use it at all).  

Feel free to write additional targets and/or scripts to your own work area based on your own workflows or tasks you frequently perform.  The github code doesn't necessarily need to have all these things in it as they are more of a personal preference depending on what you frequently work on.  

If your workflow needs to have a top level "make" target that runs lcov for only a specific app, feel free to add that to your own work area - no issues with doing that.  But it doesn't necessarily need to be part of the open source sample makefile.

asgibson, 2020-09-02:
How is it determined if something is a "personal preference" or a "feature that would be useful to the community?"

astrogeco, 2020-09-02:
> How is it determined if something is a "personal preference" or a "feature that would be useful to the community?"

Great question! If you think it's a useful feature I encourage you to write an issue.

jphickey, 2020-09-02:
> How is it determined if something is a "personal preference" or a "feature that would be useful to the community?"

We all argue about it until there is some sort of consensus?? :-)

For instance, my personal workflow involves the debugger in my preferred IDE.  I only run `make test` (top level) to get a summary of the full test set to see what needs attention (if any).  If any tests fail I then instruct my IDE's debugger to run that specific test executable (i.e. I pick the failing test program from the list of executables in the dialog box) and I have a "canned" setup that triggers a breakpoint any UT assert failure.  So by running the test program again here I'll catch the first failure with full context info (before the stack is unwound) which nets me far more useful information about what went awry than I'd ever get from a console log.

Once everything is passing successfully, I run `make lcov` to check/confirm the coverage on the code I changed, if applicable.  While this nets me a complete coverage report for the whole system, I haven't found it to be a problem to just focus on the file(s) I've actually changed and ignore the others.

So I find the two existing top level "make" targets to be perfectly adequate - But that's just my workflow, other people do things differently  ... if specialized "make" targets with the ability to do coverage on a single app is better for your flow, that's fine ... feel free to implement it!  CCB might even consider merging it if you implement it and submit a PR.  I'm not convinced that features like this would have more of a niche applicability.  But if its simple and unobtrusive enough it could be a viable merge candidate.

skliper, 2020-09-02:
> How is it determined if something is a "personal preference" or a "feature that would be useful to the community?"

Another good measure is if someone considers it useful enough to actually implement it and submit a PR.  There's still a chance it's just personal preference and may get rejected, but at least the submitter can use it locally.  There's over 350 open issues across the framework, most of which are enhancement requests if you need any ideas.

asgibson, 2020-09-03:
If/when I have a PR to offer, I will do so.  Putting in issues, before I have a concrete solution available, is just an attempt to not reinvent -- if someone already has a solution to it or proffers a better design than I have considered (which is the reason for the "Describe alternatives you've considered" part of the issue entry?), I don't want waste time coming up with something else.

astrogeco, 2020-09-03:
> If/when I have a PR to offer, I will do so. Putting in issues, before I have a concrete solution available, is just an attempt to not reinvent -- if someone already has a solution to it or proffers [...]

The issue template is a guideline, I think it's important to voice our questions and ideas even if we don't have the answers or the solutions. I don't want the template item "Describe alternatives you've considered" to discourage people from proposing ideas.

CDKnightNASA, 2020-09-03:
First, I am not all that familiar (yet) with lcov (and gcov) but I thought I'd weigh in with this discussion since it's relevant to a similar discussion I was having with Joe yesterday. Hopefully I'm not hijacking this discussion. And any suggestions for *good* documentation for lcov/gcov would be appreciated, the man pages and docs I've found are not the clearest I've read.

SBN now has 100% coverage (for the core) and I'd like to have CI fail out if it ever drops below 100%. Seems lcov really wants to focus on HTML output, which doesn't lend itself to automation of a check. I'm wondering if "make test" should, instead, be calling gcov so we could examine the results using scripts?

For cFS it might be good to also warn/err if/when the coverage % drops (by a significant %, a drop could be due to removing code that was covered; or we could just monitor total line counts and warn/err when the uncovered count goes up.)

astrogeco, 2020-09-03:
I worry about focusing too much on 100% coverage, I think it's good to be aware of it but I don't want to reject things or fail CI just because there's a drop of coverage 
https://dev.to/scottshipp/five-reasons-to-take-code-coverage-with-a-grain-of-salt-2j0

jphickey, 2020-09-03:
Yep, I concur that while 100% coverage is "nice", it doesn't really say whether the code works or not, and shouldn't be construed as evidence of such.  

To clarify - there are several tools in the mix here 

- instrumentation within the code (provided by gcc/glibc) which collects raw "arc" data, putting it into individual files for each unit.
- the `lcov` tool then collects + processes the raw data from all those units and produces another "summary" file
- the `genhtml` tool then converts that summary file from lcov into HTML form for human consumption.

If you want to get coverage stats in a script then just run the `lcov` directly, for instance, to get the coverage ratio for "sample_app" specifically one can do something like:

    $ lcov -c -d build/native/default_cpu1/apps/sample_app -o sample_app.coverage
    $ lcov --summary sample_app.coverage

This yields the following output (current bundle):

```
Reading tracefile sample_app.coverage
Summary coverage rate:
  lines......: 98.7% (149 of 151 lines)
  functions..: 100.0% (12 of 12 functions)
  branches...: no data found
```

Shouldn't be too hard to parse that output in a script if you wanted to.

CDKnightNASA, 2020-09-03:
> I worry about focusing too much on 100% coverage, I think it's good to be aware of it but I don't want to reject things or fail CI just because there's a drop of coverage
> https://dev.to/scottshipp/five-reasons-to-take-code-coverage-with-a-grain-of-salt-2j0

I agree, but it is partially a prod to make sure coverage testing is considered when adding new code. (Speaking as one who tends to *not* write test code until prodded. ;o )

skliper, 2020-09-03:
In concept I tend to disagree with not including coverage as an enforced metric for accepting contributions.  Typically a drop in coverage means the unit tests were not updated to match the code change.  This is a significant issue against a contribution, and I think it's valuable to resolve prior to merge to master.  It's not that we are trying to say coverage == functionality, it's when coverage drops it means tests are not keeping up with the code.

skliper, 2020-09-03:
One item on the checklist of our non-existent contribution process is to confirm unit tests were updated to match the code changes (both coverage and functional).  Just like not every doxygen warning is really a significant issue nor does it mean that we are keeping up with writing good documentation, but it is a minimum automated check that will catch a few cases where our code gets out of sync with the documentation.

asgibson, 2020-09-03:
>Hopefully I'm not hijacking this discussion.

No, certainly comments are welcome!  Although, I think we are all veering away from the "issue" topic and the broader testing discussion should be moved to different venue?  (In fact, it is currently going on in two places, https://github.com/nasa/osal/issues/588, my head hurts!)

>I agree, but it is partially a prod to make sure coverage testing is considered when adding new code.

The prod is not necessarily a bad thing, but 100% may not be the correct value, perhaps "if less than the last accepted percentage?" that @skliper was talking about. 

>(Speaking as one who tends to not write test code until prodded. ;o )

I certainly will aim to change your mind on this, ;-)

> ... checklist of our non-existent contribution process ...

I really like the choice of words here, made me chuckle.

CDKnightNASA, 2020-09-04:
> $ lcov -c -d build/native/default_cpu1/apps/sample_app -o sample_app.coverage
> $ lcov --summary sample_app.coverage

Thanks Joe, I've incorporated this into the SBN CI. :D

skliper, 2021-01-11:
The original request is satisfied by customizing your local Makefile test rule to include -V or -VV in the ctest call:
https://github.com/nasa/cFE/blob/1ede295c3874e6571aca965a7850d950cf7aa1a3/cmake/Makefile.sample#L137-L139

For example: `ctest -VV -O ctest.log`

Closing based on the capability to customize the behavior per project needs (project own's the Makefile).

closed 2021-01-11 by skliper