nasa/cFE #862: If a message is subscribed, then unsubscribed, additional unsubscribes do not raise error events

johnphamngc, 2020-09-03:
**Describe the bug**
If a message is subscribed, then unsubscribed, additional unsubscribes do not raise error events 

**To Reproduce**
Have SB subscribe to a message
Have SB unsubscribe to that message
Have SB unsubscribe again to that message

**Expected behavior**
Event message should be raised

**Code snips**
https://github.com/nasa/cFE/blob/08f6eab17702eac689a468974a32af8d41d4362a/fsw/cfe-core/src/sb/cfe_sb_api.c#L1059-L1112

If a message was previously subscribed to, and all pipes subscribing to it were subsequently unsubscribed, and an additional unsubscribe is issued, a valid routing index would still exist in the msgmap table, and thus an error event would not be raised.

**System observed on:**
N/A, discovered via code inspection

**Additional context**
N/A

**Reporter Info**
John N Pham, Northrop Grumman

CDKnightNASA, 2020-09-04:
This is likely a side-effect of a fix I implemented in #101

skliper, 2020-09-04:
Yeah, actually looks to me like this will seg fault (tries to access NULL pointer).  Note I'd like to add this as a test scenario to the new functional test framework (part of subscribe/unsubscribe API testing).

CDKnightNASA, 2020-09-04:
> Yeah, actually looks to me like this will seg fault (tries to access NULL pointer). Note I'd like to add this as a test scenario to the new functional test framework (part of subscribe/unsubscribe API testing).

I'm re-designing; can you point at the line you expect a seg fault?

skliper, 2020-09-04:
I don't think 1078 is true now, so 1083 will == NULL, and any of the following references to it would fault.  Didn't test, but that's what it looks like to me upon inspection.

https://github.com/nasa/cFE/blob/08f6eab17702eac689a468974a32af8d41d4362a/fsw/cfe-core/src/sb/cfe_sb_api.c#L1078-L1083

CDKnightNASA, 2020-09-04:
> I don't think 1078 is true now, so 1083 will == NULL, and any of the following references to it would fault. Didn't test, but that's what it looks like to me upon inspection.
> 
> https://github.com/nasa/cFE/blob/08f6eab17702eac689a468974a32af8d41d4362a/fsw/cfe-core/src/sb/cfe_sb_api.c#L1078-L1083

Ah, yes...I am guessing RoutePtr will always be != NULL, but ListHeadPtr could be NULL and yes line 1087 would segfault. My redesign avoids this.

PR #865 (2020-09-10): fix #862 - unsub of a message ID that is already unsubbed
Closes #862 

**Describe the contribution**
After the changes implemented in #101, there may be routing table entries with no subscribers (RoutePtr->ListHeadPtr would be NULL.) This could cause a seg-fault. Also, even if there are entries in the routing table, there will be no event generated if the unsubscribe does not find a matching route entry.

**Testing performed**
Ran unit test (with updated event count.)

**Expected behavior changes**
Repeated unsubscriptions should function fine, generating an informational event if there is no [... cut]
commit be1a23a8 (2020-09-10): fix #862 - unsub of a message ID that is already unsubbed should now work as expected
commit fd85c704 (2020-09-10): Merge pull request #865 from CDKnightNASA/fix-862-unsub_of_unsubd

fix #862 - unsub of a message ID that is already unsubbed
PR #876 (2020-09-15): Integration Candidate: 2020-09-09
**Describe the contribution**
Fix #846 
Fix #858
Fix #862 
Fix #863 
Fix #870 
Fix #873 

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/139/checks

**Expected behavior changes**
PR #872 -  Deconflict `CFE_ES_LIB_ALREADY_LOADED` and `CFE_ES_ERR_SYS_LOG_TRUNCATED` EIDs

PR #868 - Scrub all CFE references/uses of OSAL IDs to use the proper `osal_id_t` type. Any place that an OSAL ID is stored in memory or passed in an API call are changed to the `osal_id_t` type, rather than `uint32`. Conversions between this and other types (e.g. bare intege [... cut]
closed 2020-09-15 by yammajamma