nasa/cFE #873: Add unsubscribe test case to cover #862

skliper, 2020-09-09:
**Is your feature request related to a problem? Please describe.**
#862 looks like it could have resulted in a seg fault, fixed in #865
 Unsubscribe after no subscriptions are left...

**Describe the solution you'd like**
Add test case.

**Describe alternatives you've considered**
None.  Should add test to confirm code works.

**Additional context**
#862 #865 

**Requester Info**
Jacob Hageman - NASA/GSFC
@CDKnightNASA

CDKnightNASA, 2020-09-09:
Looks like current tests hits it ... lcov report shows that lines 1102-1110 were covered one time. But I will assume you mean that you'd like an explicit test.

skliper, 2020-09-10:
Note `Test_Unsubscribe_NoMatch) doesn't actually unsubscribe from MsgId or SB_UT_TLM_MID1, so it's not checking the unsubscribe followed by unsubscribe case.  The test modifies the PipeId for the MsgId description such that it won't unsubscribe.  I had to dig a bit since the test is light on comments as to the "why"...

https://github.com/nasa/cFE/blob/be1a23a8b32b6ecd03053d6924bb152d08b0c61b/fsw/cfe-core/unit-test/sb_UT.c#L2520-L2544

PR #875 (2020-09-10): Fix #873, Add test for SB subscribe/unusubscribe/unsubscribe
**Describe the contribution**
This adds a test to cover the issue in #862, and verify #865 fixed it
Fix #873

**Testing performed**
Build and ran unit test on main branch, confirmed it produced a seg fault (as expected per #860 
This should pass once #865 is checked in.

**Expected behavior changes**
None

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: bundle main + this commit

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration [... cut]
commit 8a4056fc (2020-09-10): Fix #873, Add test for SB subscribe/unsubscribe/unsubscribe
commit 9712464e (2020-09-10): Merge pull request #875 from skliper/fix873-unsub-unsub

Fix #873, Add test for SB subscribe/unusubscribe/unsubscribe
PR #876 (2020-09-15): Integration Candidate: 2020-09-09
**Describe the contribution**
Fix #846 
Fix #858
Fix #862 
Fix #863 
Fix #870 
Fix #873 

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/139/checks

**Expected behavior changes**
PR #872 -  Deconflict `CFE_ES_LIB_ALREADY_LOADED` and `CFE_ES_ERR_SYS_LOG_TRUNCATED` EIDs

PR #868 - Scrub all CFE references/uses of OSAL IDs to use the proper `osal_id_t` type. Any place that an OSAL ID is stored in memory or passed in an API call are changed to the `osal_id_t` type, rather than `uint32`. Conversions between this and other types (e.g. bare intege [... cut]
closed 2020-09-15 by yammajamma