nasa/cFE #877: CFE_MISSION_REV defined in 2 locations

skliper, 2020-09-10:
**Describe the bug**
CFE_MISSION_REVE defined in 2 locations

**To Reproduce**
Inspection

**Expected behavior**
Define in 1 location... if it's for clone and own, version seems to make sense... if it includes configuration versioning then useful in the config file.

**Code snips**
https://github.com/nasa/cFE/blob/db829294eba01b88683e679d58435f65ca02f826/cmake/sample_defs/cpu1_platform_cfg.h#L1797
https://github.com/nasa/cFE/blob/db829294eba01b88683e679d58435f65ca02f826/fsw/cfe-core/src/inc/cfe_version.h#L46

**System observed on:**
N/A

**Additional context**
None

**Reporter Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-09-11:
Need a decision on resolution approach.

jphickey, 2020-09-12:
FWIW I don't think either of these are correct - it should be in `cfe_mission_cfg.h` ...

I think the one in `cpu1_platform_cfg.h` is probably a mistake ... that file is only supposed to be platform stuff, not mission stuff.  And missions shouldn't be changing stuff directly in the CFE source code.

astrogeco, 2020-09-15:
I support it from `version.h` we'd need to apply this to all components though.

skliper, 2020-09-21:
If they are changing it though, most likely it's due to changing CFE source code already so I don't have a strong objection to just leaving it in the header file...  It's for the clone-and-own types, not really intended for configuration versioning as far as I can tell.  If it was configuration versioning, I'd expect it to be platform specific...

skliper, 2020-09-23:
CCB - 2020-09-23: remove from the config, treat this as the number that "projects" get to update as the version number.  Leave in version.h got at least 2 votes.

PR #1084 (2021-01-25): Fix #877, Remove duplicate CFE_MISSION_REV define
**Describe the contribution**
Fix #877 - removed CFE_MISSION_REV from platform config.  Intent of this rev is to track mission cFE FSW mods/customization/code changes not as a more generic "MISSION" revision number.

**Testing performed**
Nominal build and run of unit tests

**Expected behavior changes**
None, although if CFE_MISSION_REV was being utilized somewhere and included from cpu1_platform_cfg.h this will break.  Arguably this was broken before removal also since it was a duplicate define...

**System(s) tested on**
 - Hardware: cFS Dev [... cut]
commit 1b31bd99 (2021-01-25): Fix #877, Remove duplicate CFE_MISSION_REV define
commit 5b1d0549 (2021-01-25): Merge pull request #1084 from skliper/fix877-one_missionrev_2_rule_them_all

Fix #877, Remove duplicate CFE_MISSION_REV define
PR #1109 (2021-01-27): cFE Integration candidate: 2021-01-19
**Describe the contribution**

Fix #1074, Refactor CFE_TIME_RegisterSynchCallback
Fix #449, Add OS_printf to CFE_ES_SYSLOG_APPEND
Fix #488, Pad msg headers to 64 bit
Fix #903, Add CFE_SB_GetUserData padding check
Fix #877, Remove duplicate CFE_MISSION_REV define
Fix #901, Remove UT_CheckForOpenSockets references
Fix #904, Update cpuname targets.cmake documentation
Fix #1090, UT event check bounds
Fix #1052, Refactor UT_ClearForceFail to UT_ClearDefaultReturnValue
Fix #1068, Create Security Policy
Fix #933, Remove SenderReporting from SB global
Fix #985, gl [... cut]
closed 2021-01-27 by astrogeco