nasa/cFE #901: UT_CheckForOpenSockets prototype duplicated

skliper, 2020-09-22:
**Is your feature request related to a problem? Please describe.**
Prototype defined in both cFE and OSAL.

https://github.com/nasa/cFE/blob/983157db90bd205977c52762506ccbf2132837f3/fsw/cfe-core/unit-test/ut_support.h#L656-L671

https://github.com/nasa/osal/blob/f12d42ba58837a645d05eda3479d5f613ebad6c4/src/ut-stubs/utstub-helpers.h#L111-L115

Implemented here:
https://github.com/nasa/osal/blob/f12d42ba58837a645d05eda3479d5f613ebad6c4/src/ut-stubs/utstub-helpers.c#L195-L217

Also violation of magic number use in the implementation, and doesn't seem to actually do what it says (I don't see the close).

**Describe the solution you'd like**
Maybe remove if not useful?  If not, at least use the correctly scoped prototype and remove the second definition.

**Describe alternatives you've considered**
None.

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-09-23:
@jphickey - you aware of the logic behind why it was done this way?

jphickey, 2020-09-23:
> @jphickey - you aware of the logic behind why it was done this way?

This was an ugly backdoor from long ago to mimic some logic in the SB test that would confirm that all OSAL queues were indeed cleaned up by the test case.  Basically it just flags anything still open.

There is no need to close the open sockets in the current implementation.  In a very old implementation, this was required because the sockets were quasi-real entities even in UT and not closing them would interfere with future tests.  In the modern OSAL stubs just by resetting the UT state (which is done every test) you start with a clean slate every time.

Ideally there should be a better way to do this locally and only on tests that are related to cleanup.  I can investigate.

PR #1085 (2021-01-21): Fix #901, Remove UT_CheckForOpenSockets references
**Describe the contribution**
Fix #901 - removes all references to UT_CheckForOpenSockets.  No longer applicable since the UT framework resets the state for each unit test.

**Testing performed**
Built and ran unit tests, passed.

**Expected behavior changes**
None

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: Bundle main + this commit

**Additional context**
See also nasa/osal#753

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman  [... cut]
commit 483b7c00 (2021-01-21): Fix #901, Remove UT_CheckForOpenSockets references
commit c403d0e2 (2021-01-21): Merge pull request #1085 from skliper/fix901-rm-ut-checkforopensockets

Fix #901, Remove UT_CheckForOpenSockets references
PR #1109 (2021-01-27): cFE Integration candidate: 2021-01-19
**Describe the contribution**

Fix #1074, Refactor CFE_TIME_RegisterSynchCallback
Fix #449, Add OS_printf to CFE_ES_SYSLOG_APPEND
Fix #488, Pad msg headers to 64 bit
Fix #903, Add CFE_SB_GetUserData padding check
Fix #877, Remove duplicate CFE_MISSION_REV define
Fix #901, Remove UT_CheckForOpenSockets references
Fix #904, Update cpuname targets.cmake documentation
Fix #1090, UT event check bounds
Fix #1052, Refactor UT_ClearForceFail to UT_ClearDefaultReturnValue
Fix #1068, Create Security Policy
Fix #933, Remove SenderReporting from SB global
Fix #985, gl [... cut]
closed 2021-01-27 by astrogeco