nasa/cFE #910: Use CFE_SB_MessageStringSet for setting strings with two different sizes

skliper, 2020-09-23:
**Is your feature request related to a problem? Please describe.**
Anywhere that the Name, EntryPoint, or FileName members are filled in telemetry should use CFE_SB_MessageStringSet- which is similar to strncpy but takes both a source and dest length. It is intended for this use case where the local/platform and global/mission sizes might be different and this handles padding/termination issues.

**Describe the solution you'd like**
Use CFE_SB_MessageStringSet whenever sring sizes could be different.

**Describe alternatives you've considered**
Use standard patterns to avoid unterminated strings (force a NULL).

**Additional context**
#899 

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-09-30:
Upon closer inspection we may not want to use `CFE_SB_MessageStringSet()` here as it does not guarantee null termination of the output (this is intentional for fixed-size string buffers).

But for these structs we do need to ensure null termination because there are other examples of code that get these structs and then pass the strings directly to C library.

So we just need to ensure that any references to these strings use the right size, preferably using `sizeof()` rather than a specific symbol/macro name.

skliper, 2020-09-30:
Updated string copies and compares as part of #899, so this request is OBE

closed 2020-09-30 by skliper