nasa/cFE #915: cFE req. cES1007.2: Restart Application - Reject On Missing File

dmknutsen, 2020-09-28:
**Describe the bug**
cFE requirement cES1007.2 specifies that a user can not restart an Application if the original file has been removed.  If the user attempts to restart an application following removal of file, the cFE shall reject the Command, increment the invalid Command counter, and generate an event message. 

This is not what happens when the restart application command is received.  If a user has deleted the cFE Application file and then sends the restart application command.  The application successfully restarts.

**To Reproduce**
Steps to reproduce the behavior:
Delete an applications shared library file from the /cf/ location.
Send the associated restart application command.

**System observed on:**
Oracle VM VirtualBox
OS: ubuntu-19.10
Versions: cfe v6.8.0-rc1+dev28, osal v5.1.0-rc1+dev12, psp v1.4.0+dev76

**Reporter Info**
Dan Knutsen
NASA/Goddard

skliper, 2020-09-28:
Related to #424, which was reported as a requirements failure since the restart didn't work.  Need to resolve if something changed, or a different process is being used, or if it's a system dependency.

skliper, 2021-01-12:
Now that osal is working as expected, we are still failing this requirement since the command doesn't error (the request to restart is sent correctly so the valid command counter is incremented).  Likely worth just adding the file check so it is even more similar to reload.

skliper, 2021-01-12:
Actually there's 3 requirements that no longer are met in terms of incrementing the error counter:
cES1007.2 (already documented above)
cES1007.3:  If the cFE Application Restart fails due to a non-parameter error, then the cFE shall delete the Application, **increment the invalid Command counter**, and generate an event message.
cES1008.3: If the cFE Application Reload fails due to a non-parameter error, then the cFE shall delete the Application, **increment the invalid Command counter**, and generate an event message.

Command just does the request, failure to load due to a non-parameter error does not increment the invalid command counter.

skliper, 2021-01-25:
This issue is partially fixed by #1038, still will fail on missing file though without incrementing the invalid command counter.

PR #1128 (2021-02-03): Fix #915, Check for file existence in CFE_ES_RestartApp
**Describe the contribution**
Fix #915 - `CFE_ES_RestartApp` now checkes for file existence as part of command processing.  Avoids removing the app if the file doesn't exist (just avoids one error case). 

**Testing performed**
Build and ran unit tests, passed.  Also ran cFE and tested with valid and invalid file (moved good file), confirmed error counter increment (and command rejected so app remained running).

**Expected behavior changes**
Now rejects command and increments command error counter if file is missing

**System(s) tested o [... cut]
commit 79c4210d (2021-02-03): Fix #915, Check for file existence in CFE_ES_RestartApp
commit d9f3fa4a (2021-02-03): Merge pull request #1128 from skliper/fix915-restart_file_check

Fix #915, Check for file existence in CFE_ES_RestartApp
PR #1150 (2021-02-16): cFE  Integration candidate: 2021-02-02
**Describe the contribution**
Fix #1070, patch PSP include directory reference
Fix #1108, split resource id functions into module
Fix #723, CFE_EVS_Register const correct and report truncation
Fix #680, Remove table service exclusion logic
Fix #609, Remove CFE_PLATFORM_EVS_LOG_ON undefined option to diasble log
Fix #915, Check for file existence in CFE_ES_RestartApp
Fix #1122, ES and EVS use default message limits
Fix #1123, Convert global CFE_SB_Default_Qos to macro
Fix #780, Hex output for return checking ASSERTs
Fix #986, Remove unnecessary CFE_MSG_Ini [... cut]
closed 2021-02-16 by astrogeco