nasa/cFE #923: Repetitive AppID lookups in CFE_SB_SendMsgFull

jphickey, 2020-09-30:
**Describe the bug**
SB has some code to filter an app's own messages via `CFE_SB_PIPEOPTS_IGNOREMINE`.   This gets the appid and compares to the AppID of the pipe creator, and skips the destination if its a match:

https://github.com/nasa/cFE/blob/8a7dc8f2e434889287973c41d4a5cb4a48f48b44/fsw/cfe-core/src/sb/cfe_sb_api.c#L1359-L1369

The problem is that this is "inside the loop" of all destinations in the routing entry.  So it will (potentially) call `CFE_ES_GetAppId()` multiple times.  This also creates a double locking situation, because the SB lock is being held at the time this executes, and the ES lock needs to be acquired by `CFE_ES_GetAppId()`

**Expected behavior**
Code should query the caller AppID early, before taking the SB lock.

**Additional context**
The code works but is inefficient, and double locking is a potential deadlock.

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-09-30:
I'm calling this a bug even though it runs due to the double lock and that it is a potential code standards issue - as (I think?) we are not supposed to create local vars inside the function code, only at the beginning?

astrogeco, 2020-09-30:
> I'm calling this a bug even though it runs due to the double lock and that it is a potential code standards issue - as (I think?) we are not supposed to create local vars inside the function code, only at the beginning?

Do you have access to the style guide? Funny enough I stumbled on to this stack overflow [discussion](https://stackoverflow.com/questions/288441/variable-declaration-placement-in-c#:~:text=%20From%20a%20maintainability%2C%20rather%20than%20syntactic%2C%20standpoint%2C,variables%20at%20the%20beginning%20of%20t...%20More%20) that discussed it. I'll need to go through the code standard and see where it mentions it.

skliper, 2020-09-30:
3.7 - (6) "Function scope variable declarations shall precede any code statements. "

jphickey, 2020-09-30:
There are other examples of block-scope variables being used, such as here:

https://github.com/nasa/cFE/blob/8a7dc8f2e434889287973c41d4a5cb4a48f48b44/fsw/cfe-core/src/sb/cfe_sb_api.c#L1397

To clarify this is being declared in a _block_ so it has block scope, not in a function scope. And they are at least at the beginning of the block, before code in that block.

Still, I think it is best to avoid the practice when possible, I consider it almost the same thing....

PR #925 (2020-10-01): Fix #923, Perform appid lookup early
**Describe the contribution**

Move the AppID lookup to be early in the CFE_SB_SendMsgFull implementation. Avoids double locking between SB+ES and avoids a block-scope local variable.

Fixes #923 

**Testing performed**
Build and sanity check CFE
Run unit tests

**Expected behavior changes**
None.

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Joseph Hickey, Vantage Systems, Inc.
commit 948fa07b (2020-10-01): Merge pull request #925 from jphickey/fix-923-appid-lookups

Fix #923, Perform appid lookup early
commit a4ffe5b5 (2020-10-01): Fix #923, Perform appid lookup early

Move the AppID lookup to be early in the CFE_SB_SendMsgFull implementation.
Avoids double locking between SB+ES and avoids a block-scope local variable.
PR #936 (2020-10-06): Integration Candidate 2020-10-07
**Describe the contribution**
Fix #923 
Fix #897  
Fix #651
Fix #923
Fix #705
Fix #888

**Testing performed**
See Bundle: https://github.com/nasa/cFS/pull/145/checks

**Expected behavior changes**

PR #898 - Adds the field `UnregAppID` to track whether an "unregistered" event was generated, un-overloading the `EventCount` field to serve its primary purpose of counting actual events generated from a valid/registered AppID.

PR #925 - Move the AppID lookup execution to be early in the CFE_SB_SendMsgFull implementation. This avoids double locking between SB+ES and [... cut]
closed 2020-10-06 by astrogeco