nasa/cFE #927: Stub for CFE_ES_RunLoop default return is false, should it be?

asgibson, 2020-09-30:
**Describe the bug**
CFE_ES_RunLoop uses UT_DEFAULT_IMPL which typically returns CFE_SUCCESS, which happens to also equal 0.  CFE_ES_RunLoop returns UT_DEFAULT_IMPL !=0, which causes the default return to be false (0 != 0).  Is that the desired default behavior?

**To Reproduce**
Steps to reproduce the behavior:
1. Write a unit test that expects CFE_ES_RunLoop to succeed by default.
2. Run test, see that is not the behavior.

**Expected behavior**
I had expected 'true' to be the default.

**Code snips**
https://github.com/nasa/cFE/blob/8a7dc8f2e434889287973c41d4a5cb4a48f48b44/fsw/cfe-core/ut-stubs/ut_es_stubs.c#L1085-L1089

To run a single loop for a unit test, this is required:
`UT_SetForceFail(UT_KEY(CFE_ES_RunLoop), true);`
`UT_SetDeferredRetcode(UT_KEY(CFE_ES_RunLoop), 2, false);`

The main reason this requirement does not make sense is UT_SetForceFail makes one think it should fail, not succeed.

**System observed on:**
RHEL 7.6

**Additional context**
If this is the desired behavior, close and disregard this issue.

**Reporter Info**
Alan Gibson  NASA GSFC/587

jphickey, 2020-09-30:
Yes, to default to `false` is indeed desired behavior, because in the typical use-case this function is the conditional on a `while` loop.  So if defaulted to `true` one would have an infinite loop if no return code was configured.

Defaulting to false is just an easy way to keep it simple and avoid an infinite loop if the test case forgets to explicitly configure something.  If you want one loop you should be able to just do:

`UT_SetDeferredRetcode(UT_KEY(CFE_ES_RunLoop), 1, true)`

And nothing else, don't need to bother with "force fail" unless you want many iterations.

skliper, 2020-09-30:
Also related to nasa/OSAL#559 (UT_SetForceFail misnomer).

asgibson, 2020-09-30:
The single loop was an example, so yes many iterations will require the force fail.

Where is the behavior for CFE_ES_RunLoop stub defined outside of the code?  Do all bool return functions for stubs have this same behavior or just this one because of the loop usage?