nasa/cFE #928: Move SB route lookup (including insert method) to a module

skliper, 2020-09-30:
**Is your feature request related to a problem? Please describe.**
Various implementations possible for message and route tables, all with associated advantages and issues.

**Describe the solution you'd like**
Provide the capability to replace the core implementation.

**Describe alternatives you've considered**
None.

**Additional context**
For implementations that support large MsgIds, or prefer smaller memory footprint at the cost of performance.  Hashes, searches, etc.

**Requester Info**
Jacob Hageman - NASA/GSFC (per stakeholder request for alternate implementation)

skliper, 2020-10-05:
Abstracting out MsgId/Route relation proposal - CFE_SB_GetRoutingTblPtr could use MsgId, with all the relation logic underneath the hood.  CFE_SB_GetDestPtr could use route pointer.  For the most part hide CFE_SB_GetRoutingTblIdx... shouldn't use direct access except when looping for reporting, and then could just loop the table.  Also move all the related utility functions.  Replace CFE_SB_RouteIdxPop_Unsync with a generic CFE_SB_AddRoute given MsgId, and CFE_SB_RouteIdxPush_Unsync is dead code for now (we no longer remove routes).  CFE_SB_IsValidRouteIdx may actually go away (don't expose index directly).

cfe_sb_task.c
- CFE_SB_EnableRouteCmd - get route, get dest 
- CFE_SB_DisableRouteCmd - get route, get dest
- CFE_SB_SendRtgInfo - loop through routes, not keys
- CFE_SB_SendMapInfo - loop through routes, not keys

cfe_sb_priv.c
 - CFE_SB_DuplicateSubscribeCheck - use route pointer (used in one place, could get route first)

cfe_sb_init.c
 - CFE_SB_InitMsgMap - TBD, at minimum move into module

cfe_sb_api.c - straight forward using same methods above, use MsgId to get route pointer..

skliper, 2020-10-05:
Not planning on fixing with this change - PipeId is used as an index in multiple places, whitespace/alignment is tragic, excessive comments (in my opinion).  Likely spawn follow-on PRs to clean up.

jphickey, 2020-10-05:
Old issue #100 is for making Pipe IDs abstract/safe (which must include fixing use as index, etc).

Might be worth considering for this cycle?  Since I've converted so many other resources to use abstract IDs already, seems like a natural thing to include.  Could be done after the current efforts are completed, if desired.

skliper, 2020-10-06:
> Could be done after the current efforts are completed, if desired.

Low priority but likely possible, and a desirable change for consistency.

astrogeco, 2020-10-07:
[CCB 2020-10-06](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-October-06-2020) 

Two implementation options:

- Option 1: Hard option, doing it "the right way"
  - Redesign the interface using a new module
  - Abstract the relationship between routes and message IDs

- Option 2: Easy option
  - Small fixes to the current design if we don't make it "interface safe"
  - Problems with maintenance if the routing table changes

Votes for the structured API method (Option 1)

skliper, 2020-10-07:
I think the key to option 1 is to keep direct access to the resource (routing table in this case) on one side of the interface.  Option 2 was allowing direct access on both sides in the short term, since that's how it's done currently since today routing is an internal capability of SB.  Making routing a module with various possible implementations benefits from a more defined interface between routing and SB, which is option 1.  Even doing it the right way shouldn't be that much more work, I'm just lazy enough to want confirmation that the harder option is worth it from the CCB/open source point of view.

PR #947 (2020-10-21): Fix #928 and #929 - Modularize software bus routing, add msg map hash
**Describe the contribution**
Fix #928 - refactor SB and move routing to a module
Fix #929 - adds message map hash implementation
Partial #948 - fixed double lock potential

Key concepts:
 - added SBR module
 - Message map and routing table are in SBR, access APIs on SB side
 - SB still owns destination logic
 - no more route index or pointers being passed around, all route id and message id based
 - hash message map "oversized" (4x) to minimize collisions (only ~10% single collisions on the real system with a full routing t [... cut]
commit 0aac36a1 (2020-10-21): Fix #928, Added software bus routing module

- Implementation for direct map and unsorted routing table
- Includes full coverage tests
- Removed msg key and route stack concepts from direct map
commit 12f2ff1b (2020-10-21): Merge pull request #947 from skliper/fix928-route-direct-module

Fix #928 and #929 - Modularize software bus routing, add msg map hash
commit 33364e81 (2020-10-21): Fix #928, Extracted msg map and route table from SB

- Removed all index/pointer accesses of message map and routing table
- All references now via IDs and APIs
- Note SB still owns destination logic (unchanged linear linked list)
- Limited whitespace fixes for readability
- Resolved observed instances of variables not declared at the
  start of functions
- Cleaned comments
- Resolved potential double locks in CFE_SB_SendPrevSubs
- Route and message write to file no longer guaranteed in msgid
  order to maintain performance for large msgid space implementations
- Removed unused CFE_SB_FindGlob [... cut]
closed 2020-10-28 by astrogeco