nasa/cFE #945: CFE_PLATFORM_ES_PERF_MAX_IDS not fully deprecated

pchapates, 2020-10-13:
**Describe the bug**
CFE_PLATFORM_ES_PERF_MAX_IDS was superseded by CFE_MISSION_ES_PERF_MAX_IDS as noted in this comment: https://github.com/nasa/cFE/search?q=CFE_PLATFORM_ES_PERF_MAX_IDS. However, sample cpu1_platform_cfg.h still contains the definition for CFE_PLATFORM_ES_PERF_MAX_IDS is still referenced in es_UT.c and comments in cfe_es_events.h and sample_perfids.h

**To Reproduce**
N/A

**Expected behavior**
Either CFE_PLATFORM_ES_PERF_MAX_IDS should be totally deprecated and all references should be replaced by CFE_MISSION_ES_PERF_MAX_IDS or (if deemed necessary) support for platform-specific max values should be re-added in the perf-log implementation.

**Code snips**
cfe/cmake/sample_defs/cpu1_platform_cfg.h:1978
cfe/fsw/cfe-core/src/inc/cfe_es_events.h:1046
cfe/fsw/cfe-core/unit-test/es_UT.c:3664

**System observed on:**
N/A

**Additional context**
N/A

**Reporter Info**
PJ Chapates Gateway Vehicle System Manager FSW Production, JSC

jphickey, 2020-10-13:
CCB topic - 

For this we need to decide if its worth having a (potentially) lower PLATFORM limit vs the MISSION limit.   For instance, one can set `CFE_PLATFORM_ES_MAX_APPLICATIONS` lower than `CFE_MISSION_ES_MAX_APPLICATIONS` which can save memory on a local processor while still keeping telemetry packet sizes consistent across all CPUs.  There may be a valid reason to implement a similar thing for perfids too.

That will determine whether this goes away completely or it gets reinstated as the size of internal tables.

skliper, 2020-10-14:
CCB: 2020-10-14 - Discussion: mission sizes telemetry, the platform version could size memory footprint.  This isn't a requirement at this point so the path forward is to just remove the platform defines, open an optional ticket that covers the potential to reduce memory footprint (not targeted to a build.)

skliper, 2021-01-19:
> [CCB: 2020-10-14](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-October-14-2020) - Discussion: mission sizes telemetry, the platform version could size memory footprint. This isn't a [...]

See #1103 for the new issue

PR #1104 (2021-01-25): Fix #945, Finish CFE_PLATFORM_ES_PERF_MAX_IDS removal
**Describe the contribution**
Fix #945 - Removes the rest of the references/uses of `CFE_PLATFORM_ES_PERF_MAX_IDS` (in favor of `CFE_MISSION_ES_PERF_MAX_IDS `)

**Testing performed**
Build and run unit tests, passed

**Expected behavior changes**
None, just avoids issues that could happen if the platform/mission defines diverged

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: Bundle main + this commit

**Additional context**
#1103 is open for consideration, but future work

**Third party code**
None

** [... cut]
commit 21bf38a4 (2021-01-25): Fix #945, Finish CFE_PLATFORM_ES_PERF_MAX_IDS removal
commit cc514bc6 (2021-01-25): Merge pull request #1104 from skliper/fix945-platform_perf_max_id

Fix #945, Finish CFE_PLATFORM_ES_PERF_MAX_IDS removal
PR #1109 (2021-01-27): cFE Integration candidate: 2021-01-19
**Describe the contribution**

Fix #1074, Refactor CFE_TIME_RegisterSynchCallback
Fix #449, Add OS_printf to CFE_ES_SYSLOG_APPEND
Fix #488, Pad msg headers to 64 bit
Fix #903, Add CFE_SB_GetUserData padding check
Fix #877, Remove duplicate CFE_MISSION_REV define
Fix #901, Remove UT_CheckForOpenSockets references
Fix #904, Update cpuname targets.cmake documentation
Fix #1090, UT event check bounds
Fix #1052, Refactor UT_ClearForceFail to UT_ClearDefaultReturnValue
Fix #1068, Create Security Policy
Fix #933, Remove SenderReporting from SB global
Fix #985, gl [... cut]
closed 2021-01-27 by astrogeco