nasa/cFE #949: Use memchr() for string length checks

jphickey, 2020-10-15:
**Is your feature request related to a problem? Please describe.**
String length verification checks are generally done via the `strlen()` C library API.  The potential issue here is that `strlen` requires/assumes a termination byte to be present.  If the user mistakenly passes an unterminated string, the implementation will search sequentially in memory beyond the string (i.e. forever) until it either encounters a NUL char somewhere in memory, or segfaults, whatever comes first.

**Describe the solution you'd like**
Use `memchr()` for string length check instead, which can specifically check for a NUL char within a certain length.  For instance, instead of:

```
if (strlen(string) >=  MAX_LENGTH)
{
   /* string too long.... */
}
```

use:

```
if (memchr(string, 0, MAX_LENGTH) == NULL)
{
   /* string too long.... */
}
```

This has the advantage that it will _not_ search beyond the specified `MAX_LENGTH`, so its a bit safer.  It's also a more direct check - the code generally doesn't care what the actual string length is, it just needs to know if its within MAX_LENGTH or not.

**Describe alternatives you've considered**
Leave as is.

**Additional context**
This is a minor thing, but would improve robustness a bit.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

CDKnightNASA, 2020-10-15:
I like! But how will folks find out they forgot to null-terminate if we don't crash? ;)

jphickey, 2020-10-15:
> I like! But how will folks find out they forgot to null-terminate if we don't crash? ;)

It'll probably still segfault later, because the function returns an error code that they aren't checking either :-)