nasa/cFE #965: Avoid OSAL IDs in messages/data files

jphickey, 2020-10-27:
**Is your feature request related to a problem? Please describe.**
OSAL runtime types aren't really defined as part of the external (i.e. mission-scope) interface.  The `osal_id_t` is one such example - it is a local runtime type, not really intended to be saved in data files or command/telemetry messages.

This might seem pedantic because it is simply a `uint32`, but when using CFE with a command/data dictionary tool this becomes apparent that the OSAL types used in telemetry and data files aren't part of the data dictionary.  (OSAL itself doesn't have a cmd/tlm interface so it naturally wouldn't provide any such entity).

**Describe the solution you'd like**
Use the `CFE_ES_ResourceID_t` instead.  Call `CFE_ES_ResourceID_FromOSAL()` when writing and `CFE_ES_ResourceID_ToOSAL()` when reading, to do the type conversion.  The underlying value is compatible (i.e. both 32 bit `uint32`, same numbers) so it should be transparent to external tools.

**Describe alternatives you've considered**
Have OSAL provide a stablized CMD/TLM definition of IDs?  (but CMD/TLM interface is not really part of OSAL's role - it is a runtime library)

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

astrogeco, 2020-10-28:
[CCB 2020-10-28](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-October-28-2020) 
- Consider removing OSAL IDs in dump files since they're not consistent and are runtime specific
- @jphickey to make list of where the IDs pop up in the codebase

jphickey, 2020-10-30:
Checked the CFE code, I only see one instance of OSAL IDs in messages or data files - the `ModuleId` field of the `CFE_ES_AppInfo` struct which in turn is reported within the "One App" telemetry message as well as the "query all apps" data file:

https://github.com/nasa/cFE/blob/e0d1ce883476931c86adc00adac49edea1a30d5a/fsw/cfe-core/src/inc/cfe_es.h#L363

So this currently (at least out of the box) probably has no meaning or usefulness for a ground system.  

I said "currently" because although AFAIK there is no interface that allows a ground system to query OSAL resources by ID, someone could easily write an app to do so, in which case the ID becomes useful.

The other option would be to just switch it to the `CFE_ES_ResourceID_t` type instead, which is backward compatible.

jphickey, 2020-10-30:
Found another one, the `CFE_SB_PipeD_t` structure is written in the SendPipeInfo command.  This one had thrown me off because it appears in the `cfe_sb_priv.h` file - suggesting it is an internal data structure.  But it actually is written to a stats file, making it effectively also a telemetry message.  This is probably a separate issue to resolve...

jphickey, 2020-10-30:
Submitted #982 about "private" SB Pipe data

commit 720525ed (2020-12-01): Fix #965, remove OSAL ID from App/LibInfo struct

The "osal_id_t" type isn't defined in any of the CFE message/interface
header files for use within telemetry.

This ID is an ephemeral runtime value and is not relevant to a ground
system or anything outside CFE, so it makes most sense to simply
remove this ID from the telemtry structure.  Note that all commands
are name-based, not ID-based, hence why this ID is not really useful.
commit c4950a68 (2020-12-01): Merge pull request #1017 from jphickey/fix-965-osal-ids-tlm

Fix #965, remove OSAL ID from App/LibInfo struct
PR #1017 (2020-12-01): Fix #965, remove OSAL ID from App/LibInfo struct
**Describe the contribution**
Removes the Module ID from the App/Lib info telemetry structures.

Fixes #965 

**Testing performed**
Build and sanity test CFE
Run all unit tests

**Expected behavior changes**
App and Lib info telemetry structures no longer contain the "ModuleId" value from OSAL.

**System(s) tested on**
Ubuntu 20.04

**Additional context**
CFE code should be distinct about the types used in messages vs. the types used at runtime.  They may need to be different depending on mission architecture (i.e. mixing 32 and 64 bits, differe [... cut]
PR #1027 (2020-12-02): cFE Integration Candidate: 2020-11-24
**Describe the contribution**

Fix #777, Use MSG APIs
Fix #1016, bogus warning in strncpy
Fix #1004, default stack size
Fix #965, remove OSAL ID from App/LibInfo struct
Fix #1023, correct stub appid parameter
Fix #663, More consistent use of datatypes

**Testing performed**
See events for "github actions test" at <https://github.com/astrogeco/cFS/actions>

**Expected behavior changes**

PR #1003 - Backward compatible API change. Replace many uses of generic uint16 and uint32 with a more purpose-specific type. Replace all sizes with size_t across the API. 
 [... cut]
closed 2020-12-02 by astrogeco