nasa/cFE #966: CFE_ES_CDSRegDumpRec_t has implicit padding

skliper, 2020-10-27:
**Is your feature request related to a problem? Please describe.**
Related to #664 - implicit padding should be avoided

**Describe the solution you'd like**
Eventually pack definitions will come from a common definition, with explicit padding

**Describe alternatives you've considered**
None

**Additional context**
Observed in testing Bootes

**Requester Info**
Jacob Hageman - NASA/GSFC

@wmoleski

jphickey, 2020-10-27:
Actually I noticed this as part of #909 because this structure is used for data files (and therefore should be included in the msg structs as it is part of the interface) and fixed it as part of that change.

jphickey, 2020-10-27:
Checked again - there is one more thing needed here.  Previous change fixed the name length to use mission-scope definition which is a multiple of 4, fixing that part of the problem, but there is still a padding due to the use of a `bool` value (1 byte) - so the alignment does need adjustment.

commit d21d94ae (2020-10-30): Fix #966, update padding in CDS registry record

As the name field is a multiple of 4 bytes, there needs to be 3 bytes
of padding, not 1, to avoid implicit padding.

This doesn't change anything, it just makes the padding explicit instead
of implicit.
PR #991 (2020-11-04): Integration Candidate: 2020-11-03
**Describe the contribution**

Fix #909, reorganize ES public API + msg definitions
Fix #966, update padding in CDS registry record
Fix #370, rename MAX CDS name macro to FULL_NAME_LEN
Fix #976, short background task name
Fix #970, strncpy warning
Fix #988, add casts on printf calls

**Testing performed**
See framework-bundle PR: https://github.com/nasa/cFS/pull/154/checks

**Expected behavior changes**

PR #977 - Keeps task names under 16 chars to make more debugger friendly, regardless
of the OSAL limit. Task name shows up as `ES_BG_TASK`

PR #964 
- Move ES [... cut]
closed 2020-11-04 by astrogeco