nasa/cFE #979: Report task details in QUERY_ALL_TASKS command

jphickey, 2020-10-28:
**Is your feature request related to a problem? Please describe.**
The current `CFE_ES_TaskInfo` structure, which is written by the QUERY_ALL_TASKS command, only has some very basic task information - the name/ID, execution count, and what app the task is associated with.

**Describe the solution you'd like**
Suggest to also include:
- stack size
- priority
- entry point? (would depend on #978)

**Additional context**
stack size is an important statistic for determining memory usage
priority is potentially important for real time concerns
Both of these are tracked by OSAL and easily available, just have to get it and put it in the structure.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-10-28:
Question for @acudmore, @jwilmot, @skliper -- this is pretty easy to add if you concur its worthwhile.

skliper, 2020-10-29:
Good idea but not a priority.

commit 2888450a (2021-03-05): Fix #979, add stack size and priority to task info

Add the task info and priority to the data structure written by
the QUERY_ALL_TASKS command.  Leaving out entry point, as it would
be a memory address and not useful/relevant outside the process.
closed 2021-03-05 by skliper