nasa/cFE #982: SB private/internal data structure being written to telemetry dump file

jphickey, 2020-10-30:
**Describe the bug**
The `CFE_SB_PipeD_t` structure is defined within `cfe_sb_priv.h` and is internally used to manage the pipe state.

But as part of the "Send Pipe Info" command this structure is also directly written to a data dump file - making it effectively into telemetry data.

**Expected behavior**
Code should clearly delineate what is telemetry data intended to be consumed by the ground or other systems, and what is internal data that only resides in local memory.

In particular,  the `CFE_SB_PipeD_t` also contains pointers, so the dump file resulting from a Send Pipe Info command will contain these internal memory address (meaningless outside the current CPU) and it will change size depending on whether it is running on a 32-bit or 64-bit CPU.

**System observed on:**
Ubuntu 20.04

**Additional context**
Whether data is sent directly in a telemetry message in real time or dumped to a file and transferred to the ground system in a deferred fashion, it is all basically telemetry data and the structures used in these data dump files should be defined in consistent (i.e. non platform dependent) terms so the tools on the ground can actually parse the file (or message).

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-10-30:
My suggestion/recommendation is to make this more like what ES does with its AppInfo/TaskInfo structures.  Instead of directly writing the global data records i.e. `CFE_ES_AppRecord_t` / `CFE_ES_TaskInfo_t`, it defines a separate `CFE_ES_AppInfo_t` and `CFE_ES_TaskInfo_t` which are used for data dump and telemetry packets.  These have mostly the same information but the latter is "sanitized" to make it into a consistent binary format.

skliper, 2020-11-02:
This would be a good time to fix.  Concur w/ recommendation.

skliper, 2020-11-03:
Consider along with #995 and possible inclusion in 6.8 rc.

PR #1102 (2021-01-26): Fix #982, separate pipeinfo file data structure
**Describe the contribution**
Provide a separate/dedicated structure for use in the "Pipe Info" file as written by the `CFE_SB_SEND_PIPE_INFO_CC` command, instead of using the `CFE_SB_PipeD_t` which is supposed to be internal/private.

Data is extracted from the internal `CFE_SB_PipeD_t` while shared data locked and staged into this new format, then the shared data is unlocked while the other information is gathered and finally written to the file.

Fixes #982 
Fixes #995 

**Testing performed**
Build CFE and sanity check, run all unit tests
Run  [... cut]
commit 6d40f296 (2021-01-26): Fix #982, use separate pipe info data struct

Define a data structure in cfe_sb_msg.h that will be used
with the "write pipe info" command (CFE_SB_SEND_PIPE_INFO_CC).

This allows the internal CFE_SB_PipeD_t descriptor object to
evolve as needed without affecting the binary format of the
file that is generated form this command, and items such
as memory pointers may be excluded from the file.
commit 7d0845c9 (2021-01-26): Merge pull request #1102 from jphickey/fix-982-separate-pipeinfo

Fix #982, separate pipeinfo file data structure
commit 137768d1 (2021-01-26): Fix #982, use separate pipe info data struct

Define a data structure in cfe_sb_msg.h that will be used
with the "write pipe info" command (CFE_SB_SEND_PIPE_INFO_CC).

This allows the internal CFE_SB_PipeD_t descriptor object to
evolve as needed without affecting the binary format of the
file that is generated form this command, and items such
as memory pointers may be excluded from the file.
closed 2021-01-27 by astrogeco