nasa/cFE #992: Automatically update leap seconds in sample cmake

astrogeco, 2020-11-02:
**Is your feature request related to a problem? Please describe.**
Based on #983, it would be useful to set up a github bot to check and update this every six months or so. 

**Describe the solution you'd like**
Automate updating this file.

**Describe alternatives you've considered**
Periodically open an issue when the leap seconds fall out of date.

**Additional context**
Official source of Leap Seconds
https://datacenter.iers.org/data/latestVersion/16_BULLETIN_C16.txt

**Requester Info**
Gerardo E. Cruz-Ortiz

jphickey, 2020-11-02:
This seems unnecessarily complex to me...

At runtime, the leap seconds value comes from the "time at tone" messages.  Especially with longer term missions, there is a high likelihood that leap seconds will be changed during active operation.  So this is really an operations issue to make sure that the time at tone messages have the right value.  The "default" value being compiled into code shouldn't matter much.

My suggestion is to make it part of our release process to check the official value with each CFE release and make the default in CFE match.  But by definition these values are always changing so there should be no assumption by users that the default value provided  in the sample file is usable.  

Again - ultimately the systems engineers deploying this software will have to make sure their "time at tone" messages are sync'ed to the official leap seconds count.  The value provided as a default in the config file is only used for a brief period after a power on reset (i.e. no recovery of time values from reset area) until the first real time at tone message arrives.

astrogeco, 2020-11-02:
I thought it would be a fun project for a contributors. Agreed that at the end of the day it's up to each mission to set this up.