nasa/cFE #993: CFE_TIME_ADD_1HZ_ADJUSTMENT_CC and CFE_TIME_SUB_1HZ_ADJUSTMENT_CC commands are not performing as expected

leenayurkova, 2020-11-02:
**Describe the bug**
Add Delta to Spacecraft Time Correlation Factor each 1Hz and Subtract Delta from Spacecraft Time Correlation Factor commands are modifying STFC seconds as not as required.  

[https://etdjira.gsfc.nasa.gov/browse/CFE-360](url)

TIME: Delta Adjust STCF
Upon receipt of a Command to make a 'continuous' adjustment to the STCF, the cFE shall adjust the STCF each second by the Command-specified value. The commanded value is signed so a positive or negative adjustment may be made.

As example I have attached the print screen with the test #1308_09 results.  

[image]

//*
STCF seconds == 1000000
Given values for the CFE_TIME_ADD_1HZ_ADJUSTMENT_CC and CFE_TIME_SUB_1HZ_ADJUSTMENT_CC  commands are:
Seconds == 0 
Sub-seconds == 2147483648

**Reporter Info**
Anhelina Yurkova
Arctic Slope Technical Services, Inc.
NASA Goddard Space Flight Center

dmknutsen, 2020-11-02:
To provide further clarification - in the test detained above, the commanded value (sub-seconds = 2147483648) corresponds to .5 seconds.  The expectation was that post commanding the time would increase (or decrease depending on which command is used) 1.5 seconds per second.  What is shown above is that post commanding, time is increasing at a rate of .5 seconds per second.

skliper, 2020-11-03:
@acudmore - could you confirm the expected behavior of this command?  Is it intended to replace the 1 second increment or adjust relative to?  Do ops folks use it?

skliper, 2020-11-03:
@wmoleski - do you have any insight on the expectations for this command from the test perspective?

jphickey, 2020-11-03:
I haven't had a chance to try this locally - but note that these "adjust" commands are adjusting the __STCF__ value - not the time itself.   The printouts shown in the screenshot are showing STCF, not the CFE reference time.  The STCF value seems to be adjusting at the expected rate of +0.5 sec per sec so this seems correct to me.

I will run it locally when I get a chance to check it myself.

skliper, 2020-11-03:
> I haven't had a chance to try this locally - but note that these "adjust" commands are adjusting the **STCF** value - not the time itself. The printouts shown in the screenshot are showing STCF, not [...]

Concur.  @dmknutsen @leenayurkova can you show current time as well as the STCF?

dmknutsen, 2020-11-04:
That makes sense.  I just double checked and the cFE reference time is behaving as expected.  It's difficult to show with CTF output, but basically if you request the TIME Diagnostic packet - you can see Payload.CurrentUTC.Seconds increase at the expected rate of 1.5 seconds per second after commanding the 1Hz adjustment of .5 seconds.  So...I think we can go ahead and close this one.

Thanks for the clarification!

closed 2020-11-05 by skliper