nasa/cFE #1001: More user-friendly memory alignment from CFE_SB_RcvMsg

skliper, 2020-11-04:
**Is your feature request related to a problem? Please describe.**
Command with 8 byte alignment requires a memcpy by the user (AFAIK).

**Describe the solution you'd like**
Align the buffer for the user?  Any other options to make this easier?

**Describe alternatives you've considered**
None.

**Additional context**
Stakeholder request/question.

**Requester Info**
Jacob Hageman - NASA/GSFC (from stakeholder)

@excaliburt

jphickey, 2020-11-05:
The SB memory pool (like all mempools provided via the ES mempool API)  should be aligned to `CFE_ES_PoolAlign_t`.  This should automatically reflect/accommodate the strictest aligned types (i.e.  greatest of `long double`, `long long`, `void*`).

So SB buffers should already be aligned sufficiently for any data type - one should not need to `memcpy()` data that contains 64-bit data types.

If that isn't enough - e.g. if a user wants to align to the cache line size of their CPU for performance reasons - then we also have this option which can achieve greater alignment:

https://github.com/nasa/cFE/blob/9804b59eb34154b10bb2946bde7654f3d27f4b85/cmake/sample_defs/cpu1_platform_cfg.h#L799-L816

closed 2020-11-05 by skliper

jphickey, 2020-11-05:
Out of curousity - was there an actual observed case where an SB buffer wasn't aligned sufficiently?  If so then we have a bug....

In older versions of CFE this was definitely a problem.  I think it was initially addressed sometime back around CFE 6.6.0.

excaliburtb, 2020-11-05:
No, nothing observed for a recent version of cfe. I was just asking around if this was ever addressed.

jphickey, 2020-11-05:
OK thanks for clarifying!  

The main limitation right now (if it is a limitation) is that it is a platform setting - it applies to all pools (SB, TBL etc) on that platform.  There is no way to have stricter runtime alignment on only some pools but not others.  I don't see a real compelling use case for that though.