nasa/cFE #1041: Relation between cFE requirements and Unit Test implementation

djurjo, 2020-12-09:
Hi all!
I have some doubts about how tests are being implemented. In one hand if we look into a UT file, e.g. es_UT.c we can see that all the Tests there make reference to UT_Report as follows (I took one at random):
`UT_Report(__FILE__, __LINE__,
              UT_PrintfIsInHistory(UT_OSP_MESSAGES[UT_OSP_POR_OTHER]),
              "CFE_ES_SetupResetVariables",
              "Other cause reset");` 
I tried to have a deeper understanding of what is happening there but if I look for "CFE_ES_SetupResetVariables" o "SetupResetVariables" or similar I am not able to find extra information neither in cFE_FunctionalRequirements.csv nor in requirements.docx. 
I don't know if maybe I am not looking in the correct place or maybe that files are out of date; anyway I would like to know if there is any file where it is explained why to check "CFE_ES_SetupResetVariables - Other cause reset" is performed an UT_PrintfIsInHistory(...) instruction is performed and not another one. 

Thanks in advance and stay safe out there ,

skliper, 2020-12-09:
Hi @djurjo -  `fsw/cfe-core/unit-test` implements coverage tests (built with stubs).  For coverage testing we don't typically trace directly to requirements, instead they trace to the NPR 7150.2C requirement for coverage testing.  The goal of the asserts here is just to confirm the correct path in the software was covered by the proceeding test.  There is a separate set of requirements verification tests that trace directly to requirements.  We have yet to release those open source because they are transitioning from an internal implementation based on proprietary tools to utilizing open source tools.

djurjo, 2020-12-14:
Thanks a lot @skliper! I have one doubt about this yet, the requierements found at cFE_FunctionalRequirements.csv are tested somewhere then? Thanks in advance :)

skliper, 2020-12-14:
@djurjo - Yes, the cFE-FunctionalRequirements.csv are tested internally and results are reported in the version description document upon release (see https://github.com/nasa/cFS/blob/v6.7.0a/VDD-Aquila.md for an example).  The actual tests we run have not been released open source yet.

skliper, 2020-12-14:
Oh, probably also worth noting as documented in the VDD linked above, historically the requirements are only verified for the VxWorks, MCP750 setup.  If resources can be identified to cover the work, we plan to start running these tests on all three example platforms (likely starting with Caelum), or possibly sooner as part of Continuous Integration testing.

closed 2021-01-04 by skliper