nasa/cFE #1055: return value of CFE_SB_ReceiveBuffer stub does not match documentation

CDKnightNASA, 2021-01-04:
For the stock stub function, it returns the default (CFE_SUCCESS) return code but sets the message pointer to NULL. The documentation states that it defaults to return the TIMEOUT error. The logic for the stub may want to reflect this.

https://github.com/nasa/cFE/blob/56397a3a2c47c674892c14610e78263030441d73/fsw/cfe-core/ut-stubs/ut_sb_stubs.c#L450

skliper, 2021-01-04:
Probably worth making it like the other (new-ish) getters, in that it will report an error if the SB buffer has not been provided already (from the UT buffer).  Then can remain as the default success (like the other stubs).

skliper, 2021-01-26:
This has become a documentation issue (comments on stub description are incorrect).

closed 2021-06-29 by skliper