nasa/cFE #1070: add_psp_module cmake include bug

jmahoney-dev, 2021-01-06:
**Describe the bug**
When making PSP modules and calling `add_psp_module()` from the respective PSP module's CMakeLists.txt file, the necessary includes are not available and the code will fail to compile due to `#include "cfe_psp_module.h"` not being found. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a PSP module in psp/modules/<your module>
2. Create a CMakeLists.txt file in psp/modules/<your module> and add line for `add_psp_module(<your module> src/<your module c file>)`. 
2. Make a .c source file in `psp/modules/<your module>` and `#include "cfe_psp_module.h"` in it.
3. Make sure to include this psp module in target.cmake so it will be built w/ `SET(TGT1_PSP_MODULELIST <your module>)`
4. Build cFS

**Expected behavior**
If you have the proper PSP module boilerplate set up the compile error you should see is that compiler couldn't find `"cfe_psp_module.h"`

**Code snips**
I went ahead and modified cfe/cmake/arch_build.cmake function `add_cfe_module` include_directories line from
`include_directories(${MISSION_SOURCE_DIR}/psp/fsw/shared`
to
`include_directories(${MISSION_SOURCE_DIR}/psp/fsw/shared/inc`

...which fixed the problem.

**System observed on:**
 - Native Linux
 - OS: Ubuntu 18.04
 - Versions [cFE v6.8.0-rc2, OSAL v5.1.0-rc2, PSP v1.5.0-rc1]

**Reporter Info**
Joe Mahoney - LTA Research

jphickey, 2021-01-06:
I checked the history on this one - looks like this actually introduced by nasa/psp#172 which split the PSP targets into subdirectories.  As part of this the public include files for PSP got moved from `fsw/shared` to `fsw/shared/inc` but the reference in the arch_build.cmake in CFE never got updated:

https://github.com/nasa/cFE/blob/56397a3a2c47c674892c14610e78263030441d73/cmake/arch_build.cmake#L65

My recommendation:

1. Patch "bootes" RC to include the missing `/inc` subdirectory to make it match again.  It's definitely an incorrect ref here.  The CI builds don't actually depend on this, apparently.
2. Going forward as we migrate toward CMake interface libraries and target/interface-based includes rather than directory-based includes it should hopefully prevent this from happening again.  See issue #626.

FWIW, I think this is a good example of the benefit of interface libraries - it allows the PSP CMakeLists.txt file to declare where the headers actually reside, as opposed to having direct subdir paths sprinkled in other makefiles.

astrogeco, 2021-01-27:
Merged fix into `6.8.x` @jphickey can you open a PR to go into main or are you doing the "permanent" fix ?

PR #1072 (2021-01-27): Fix #1070, patch PSP include directory reference
**Describe the contribution**

The PSP header files are located in fsw/shared/inc, not fsw/shared.
This corrects the reference.

Fixes #1070

**Testing performed**
Modify local config to use "eeprom_stub" PSP module to implement eeprom functions, which reproduces the build failure - I confirmed the error with respect to `#include "cfe_psp.h"`

With this patch applied, the build succeeds again.

**Expected behavior changes**
None

**System(s) tested on**
Ubuntu 20.04

**Additional context**
This PR is based on Bootes RC2 as the bug exists there,  [... cut]
commit 6da6a1b1 (2021-01-27): Fix #1070, patch PSP include directory reference

The PSP header files are located in fsw/shared/inc, not fsw/shared.
This corrects the reference.
commit ea20cb3d (2021-01-27): Merge pull request #1072 from jphickey/fix-1070-psp-includedir

Fix #1070, patch PSP include directory reference
PR #1139 (2021-02-03): Fix #1070 for main branch (merge of 6.8.x after bookkeeping)
**Describe the contribution**
This just applies Fix #1070 to main

**Testing performed**
Leveraging testing from #1072 

**Expected behavior changes**
PSP module include path correct (should build)

**System(s) tested on**
No new testing, just a merge

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
closed 2021-02-16 by astrogeco