nasa/cFE #1078: CFE_SB_ReceiveBuffer() -- switch timeout to int32

CDKnightNASA, 2021-01-08:
**Describe the bug**
Per discussion at the Jan. 6, 2020 CCB, @jphickey suggested that the timeout parameter should be int32, not uint32. This should be changed.

See also #1063 

**Reporter Info**
Christopher.D.Knight@nasa.gov

jphickey, 2021-01-08:
CFE_SB_ReceiveBuffer() is already `int32` (see https://github.com/nasa/cFE/blob/1ede295c3874e6571aca965a7850d950cf7aa1a3/fsw/cfe-core/src/sb/cfe_sb_api.c#L1470-L1473)

But `CFE_SB_ReadBuffer` (internal function) is not.

Unless there is a good reason to "hide" OSAL definitions here (and I can't think of a good reason, because everything already depends on osapi.h anyway), I'd rather simplify and just make it so we have:

```
#define CFE_SB_PEND_FOREVER   OS_PEND
#define CFE_SB_POLL           OS_CHECK
```

... rather than redefining these values.

Right now we use the same underlying values but redefine them, so to be correct/safe the ReadQueue needs to translate.  With them defined this way we still have symbol independence (so SB can diverge from OSAL if it needs to, though I don't know why it would).  Then the ReadQueue function doesn't need to translate this value, so long as it shares the same definition.

jphickey, 2021-01-08:
Actually, just assign this one to me.  As I'm going through code looking for race conditions in issue #1073 the entire design of CFE_SB_ReadQueue() is broken - it is accessing PipeDscPtr without the SB lock.  I will replace this with something that works.

closed 2021-03-30 by skliper