nasa/cFE #1093: Inconsistent Pipe ID reporting in SB events

jphickey, 2021-01-13:
**Is your feature request related to a problem? Please describe.**
As a follow on for issue #1073 / PR #1092  - I noticed that the format strings of many SB event texts are not inconsistent, particularly with respect to Pipe IDs.  Some print the Pipe Name, whereas some just print the ID.  There is also a debug event that gets generated whenever `CFE_SB_GetPipeName()` runs, which means that for events that print the name, they actually generate _two_ events in the event log - which clutters things up.

Should be a rule of thumb that we should avoid generating more/different events in the process of generating an event - aside from being confusing it can also snowball.

**Describe the solution you'd like**
Only print the pipe IDs, not names., and use a consistent pattern/conversion (I recommend Hexadecimal/`%lx` conversion for resource IDs as it clearly reveals the table index in the lower 4 hex digits).  

Getting names at runtime is not totally trivial - There is CPU time to copy the string, and memory to store the name - and it makes all API functions that much more complex.  

But as long as the IDs are in the event text, the names can be looked up later after the fact by dumping the pipe stats to a file.  (assuming that #995 is fixed too).  It's just faster and easier and keeps the implementation simpler.

**Additional context**
The previous PR attempted to keep the event text the same as it was whenever possible, but this should be considered as a follow-on to clean up and simplify.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-01-13:
It also presents a question as to whether `CFE_SB_GetPipeName()` should generate an event at all - we don't generate debug events when things like `CFE_ES_GetAppName()` work as designed, so not sure why this is different.

skliper, 2021-01-13:
I'm OK with removing the debug event from `CFE_SB_GetPipeName`.  Pretty much a standing opinion (if a debug message is making nominal implementation more difficult, remove it).  I'd actually prefer name vs ID in error messages to avoid the two step process... I think it's worth the extra logic to simplify ops when addressing errors.

jphickey, 2021-01-13:
That's the other option I guess - always print the name in the event.  But then many functions  generate a debug event on the nominal/success case - so if these are to be the name too that is a lot more work for events that (most of the time) nobody is looking at.

Should we use IDs only for debug events, and for other non-debug events/errors use the name?

And I agree about NOT having a debug event for a successful `CFE_SB_GetPipeName()` -- this doesn't seem helpful.