nasa/cFE #1096: Possible race conditions in SB Pipe operations

jphickey, 2021-01-13:
**Describe the bug**
As part of debugging #1073 recently, it was noted that several historical SB APIs are not adequately locking the global data structures that they access.  In particular during pipe operations and subscription/unsubscription, some functions are accessing the "CFE_SB_PipeD_t" objects (pipe descriptors) in the `CFE_SB.PipeTbl` and routes in the routing table outside any SB shared data lock, or are not maintaining the lock from start to finish.  

It is therefore possible that another task can modify this data while being accessed by another task.

**To Reproduce**
Not actually observed "in the wild" - this was found by code inspection.  To manifest this would require running CFE on a multi-core system where task(s) were actively subscribing/unsubscribing or creating/deleting pipes _at the same time_.  

**Expected behavior**
All SB global data access should be protected against any possible concurrency issues.

**Code snips**
Some examples of potentially affected routines / operations in older CFE versions (these are observed in CFE 6.8, just a quick skim though - this should NOT be considered an exhaustive/complete list):  

- CFE_SB_EnableRouteCmd() / CFE_SB_DisableRouteCmd() - accesses pipes and routes with no lock
- CFE_SB_DeletePipeFull() - locks initially, but releases lock in the middle of "destptr" loop to do unsubscribe, which could possibly also release another task that also accesses the same pipe or change routing (i.e. changing the same list).
- CFE_SB_GetPipeName() - accesses PipeTbl without lock
- CFE_SB_SubscribeFull() - when sending the final "subscription report" telemetry message, this is actually stored a global, filled while locked, but actually _sent_ after unlock.  There is a possibility between the unlock and the send that another higher-priority task will do a subscribe and overwrite the message.

**System observed on:**
N/A

**Additional context**
During "steady state" operations where the routing table is not being changed, there isn't much risk, as normal sending/receiving of messages does not actually modify the data in the routing table.

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-01-13:
Note, all of these issues should be addressed by PR #1092 already.

astrogeco, 2021-01-13:
Thanks for this Joe!

closed 2021-03-30 by skliper