nasa/cFE #1097: Suppress format-truncation warnings

skliper, 2021-01-14:
**Is your feature request related to a problem? Please describe.**
It's only triggered when length limited *printf functions are used, and that's why we use them in the cFS context (to truncate as needed).

**Describe the solution you'd like**
Add `-Wno-format-truncation` to default flags, users can customize the flags as needed if they want to see these warnings.

**Describe alternatives you've considered**
Up to this point we've been implementing ways to suppress the individual warnings, but that just adds complexity where the point is to truncate when needed.

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

chillfig, 2022-03-30:
@skliper Is it a reasonable approach if I added `-Wno-format-truncation` to the code below to achieve the desired truncation warning suppression? https://github.com/nasa/cFE/blob/a6c6b65a43fceb623bbac5bfc305c68b03317982/cmake/sample_defs/arch_build_custom.cmake#L28-L37

This location make sense to me since

jphickey, 2022-03-31:
I can confirm that (because I'm using Ubuntu 21.10 which has GCC 11) that I have added these option to my local `arch_build_custom.cmake` file and CFE/OSAL builds fine after this.

I recommend adding both of these options,  as they produce similar nuisance warnings:

     -Wno-format-truncation
     -Wno-stringop-truncation

One affects printf-style formatting, the other affects other string ops.

jphickey, 2022-03-31:
Main thing is to check/confirm that this option is benign to older GCC versions.  I think we are still targeting/supporting CentOS7 which has gcc 4.8 (ancient history).  So it would be wise to make sure this old compiler doesn't complain about that option.

astrogeco, 2022-04-12:
> 

We should consider how many compilers we want to support and maybe draw a line just like we do with OSs and custom 
configs

commit a39d0f11 (2022-05-02): Merge pull request #2078 from chillfig/supress_trunc_warnings

Fix #1097, adds truncation warning suppression flags
commit c74da481 (2022-05-02): Fix #1097, adds truncation warning suppression flags
commit c6a00d08 (2022-05-02): Merge pull request #2078 from chillfig/supress_trunc_warnings

Fix #1097, adds truncation warning suppression flags
PR #2078 (2022-05-19): Fix #1097, adds truncation warning suppression flags
**Describe the contribution**
- Fix #1097 

Adds printf-style format and string operation truncation warning suppression flags.

**Testing performed**
Tested on fork: https://github.com/chillfig/cFE/actions?query=branch%3Asupress_trunc_warnings

**Expected behavior changes**
None, flags should prevent warnings on future compilations using latest systems. 

**System(s) tested on**
Ubuntu 18.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Justin Figueroa, ASRC Federal

EDIT - JH: Added reference to issue s [... cut]
closed 2022-05-19 by astrogeco