nasa/cFE #1117: CFE_SB_SND_RTG_EID and CFE_SB_SND_RTG_ERR1_EID used for all three write file cmds (route, map, pipe)

skliper, 2021-01-22:
**Is your feature request related to a problem? Please describe.**
Event ID's should be unique, these aren't.

**Describe the solution you'd like**
Define event ID's for all three.  Also worth a general scrub of event ID's to ensure they are unique.  Also "SND" is a misnomer, these are write commands.

**Describe alternatives you've considered**
None

**Additional context**
None.

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2023-02-03:
Note: these were re-arranged (put into the background) in https://github.com/nasa/cFE/pull/1148
These events are now grouped together in `CFE_SB_BackgroundFileEventHandler()`.
Making these EIDs unique would now require refactor of `CFE_SB_BackgroundFileEventHandler()` making it more complex and/or longer.

Perhaps just this PR should just be closed without action, or closed with a simple renaming of `CFE_SB_SND_RTG_EID` and `CFE_SB_SND_RTG_ERR1_EID` to make their names clearer.

skliper, 2023-02-03:
@thnkslprpt - Agree w/ the simple rename.  Single use in code is the guideline, and it's fine in a utility function to use the same EID from different callers.  My preference is the ability to trace from an event to a single line in code.