nasa/cFE #1119: Calculate crc can't return error code

zanzaben, 2021-01-22:
**Is your feature request related to a problem? Please describe.**
Calculate CRC returns a uint so any error code you use gets overloaded. 

**Describe the solution you'd like**
be able to handle errors 

**Additional context**
Was found while working on #547 

**Requester Info**
Alex Campbell GSFC

skliper, 2021-01-22:
Issue - a buggy app could call this API with a NULL pointer and cause a segfault.  Better to have an API that returns status, and sets the CRC as an output parameter.

jphickey, 2021-01-22:
IMO This falls into the category of where the checks themselves are adding complexity that otherwise wouldn't exist.  A "buggy app" is  unlikely to call this function with NULL, it would more likely call it with some random pointer which is not NULL but not valid either.    So a "safety check" will not catch it.  

But since nobody expects or would anticipate something like a CRC could ever fail (it is, after all, a mathematical algorithm applied to a bunch of numbers) the extra complexity of adding a int32 return code and actually expecting users to _check_ that code is just a lot of extra work for stuff that should never actually happen.

If coding standards actually require that all functions accept a NULL pointer without segfault, then I'd propose just returning the CRC init value, e.g.:

    if (DataPtr == NULL || DataLength == 0)
    {
        return InputCRC;
    }

That way the function just becomes a no-op if passed with a NULL pointer.  I still say that this is only likely to "defer" a failure and make it more complicated to debug - but if coding standards require the function to accept NULL this is an easy way to do it and does not change API or make it unnecessarily complicated.

skliper, 2021-01-22:
I'm fine w/ the suggestion above (make it a no-op, return InputCRC).  No API change is good for @ejtimmon...

zanzaben, 2021-01-25:
A similar issue happens with CFE_SB_GetChecksum. 

https://github.com/nasa/cFE/blob/c7363c8d0bfbc0ce8d404799c64defe3cb171616/fsw/cfe-core/src/sb/cfe_sb_util.c#L231-L252

currently if you send a null pointer, CFE_MSG_GetHasSecondaryHeader will catch it leaving hassechdr false and returning 0.

skliper, 2021-01-25:
CFE_SB_GetChecksum is deprecated.  Don't need to worry about it.

PR #1140 (2021-03-02): Fix #546 #547, api argument validation
**Describe the contribution**
Fixes #546 
Fixes #547
Add validation for method parameters. Mostly null pointer checks for #547 and a few size checks for #546

Fixes #1119
Returning the input instead of an error code 

**Testing performed**
Build and run unit test

**Expected behavior changes**
No impact to behavior

**System(s) tested on**
Ubuntu 20.04

**Additional Context** 
Here is the function list from #546 and the changes done 

cfe_es_api.c:CFE_ES_DeleteApp - Can get a segmentation fault if user tries to delete an APP greater than CFE_PLATFORM_ES_M [... cut]
PR #1196 (2021-03-04): cFE Integration candidate: 2021-03-02
**Describe the contribution**
Fix #1020, refactor SB buffer descriptor object 
Fix #922, Update `cfe_error.h` to use the `CFE_Status_t` type
Fix #1149, remove cfeesugshellsrv from user guide
Fix #546 , APIs missing argument validation
Fix #547 , APIs missing check for NULL pointer(s)
Fix #1119 ,  Calculate crc can't return error code
Fix #1153, Remove logic based on LogEnabled status
Fix #1189, correct compiler warnings 

**Testing performed**
See <https://github.com/nasa/cFS/pull/204/checks>

**Expected behavior changes**

PR #1154 - Refactors the SB buff [... cut]
closed 2021-03-04 by astrogeco