nasa/cFE #1123: CFE_SB_Default_Qos exposed globally, with internal defaults, passing structure by value

skliper, 2021-01-25:
**Is your feature request related to a problem? Please describe.**
CFE_SB_Default_Qos is extern from cfe_sb.h (so any code including cfe_sb.h could change it's value!):
https://github.com/nasa/cFE/blob/c7363c8d0bfbc0ce8d404799c64defe3cb171616/fsw/cfe-core/src/inc/cfe_sb.h#L175-L185

But the defines are "internal" to SB:
https://github.com/nasa/cFE/blob/c7363c8d0bfbc0ce8d404799c64defe3cb171616/fsw/cfe-core/src/sb/cfe_sb_priv.h#L86-L87

Also structure passed by value:
https://github.com/nasa/cFE/blob/c7363c8d0bfbc0ce8d404799c64defe3cb171616/fsw/cfe-core/src/inc/cfe_sb.h#L381

There is no underlying implementation, so currently just a placeholder in the API.

**Describe the solution you'd like**
Possibly convert to bits in a uint32/16 or similar (structure is overkill), provide default and the other values publicly as defines, don't expose as a global variable.

**Describe alternatives you've considered**
None

**Additional context**
Found when working #1036, it's out of family since it's not at task global scope.

**Requester Info**
Jacob Hageman

skliper, 2021-01-25:
@ejtimmon Note CFE_SB_Default_Qos is referenced by ds and hs (and in sch test utils)

jphickey, 2021-01-25:
Passing by value is generally OK so long as the structure is small (i.e. 32 bits or less) in case the calling convention in use relies on register(s) to pass argument/return values.  However if it grows larger it can become inefficient to pass by value.

IIRC the one I was worried about before was `CFE_TIME_SysTime_t` - this is passed by value and it is 64 bits in size, therefore could be inefficient on 32-bit machines depending on the calling convention and particularly the return value.

But this one is only 16 bits so its probably fine.  But it would be a good idea to note in the structure/declaration comments that it is passed by value and therefore needs to be kept small for this reason - so fields shouldn't be added without careful consideration.

skliper, 2021-01-25:
I understand it's OK in this case since it's small, but the extra caution/complexity and currently implementation really doesn't seem justifiable based on a lack of implementation.  I'd think a generic uint and setting bits is a more flexible, future proof implementation that needs no special note.  Could easily add implementations as modules and assign more bits (or have reserved/user defined bits), etc without having to redefine the elements in the structure (until bits run out).

jphickey, 2021-01-25:
Sure, bit fields in an integer could work too - but then you lose the advantage of type safety protection from having it as a struct.  That is, one has to declare their local variable as a `CFE_SB_Qos_t` or else they get a type mismatch when they try to pass it to the API.  If it is `uint32` then any `uint32` will be accepted - regardless of whether it is actually a match or not.

I definitely agree with moving the constants out of the "private" file and into a public header file - but I would hesitate to actually change the API at this time.

PR #1125 (2021-02-08): Fix #1123, Convert global CFE_SB_Default_Qos to macro
**Describe the contribution**
Fix #1123 - deprecates CFE_SB_Default_Qos and replaces with CFE_SB_DEFAULT_QOS macro that avoids global variable exposure, and removed SB internal defines (they aren't implemented anywhere and are not used).  These can be added at the appropriate scope once requirements are defined.

**Testing performed**
Build and ran unit tests

**Expected behavior changes**
None

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: Bundle integration candidate + this commit

**Additional conte [... cut]
PR #1150 (2021-02-16): cFE  Integration candidate: 2021-02-02
**Describe the contribution**
Fix #1070, patch PSP include directory reference
Fix #1108, split resource id functions into module
Fix #723, CFE_EVS_Register const correct and report truncation
Fix #680, Remove table service exclusion logic
Fix #609, Remove CFE_PLATFORM_EVS_LOG_ON undefined option to diasble log
Fix #915, Check for file existence in CFE_ES_RestartApp
Fix #1122, ES and EVS use default message limits
Fix #1123, Convert global CFE_SB_Default_Qos to macro
Fix #780, Hex output for return checking ASSERTs
Fix #986, Remove unnecessary CFE_MSG_Ini [... cut]
commit 7a2dcce8 (2021-02-16): Fix #1123, Convert global CFE_SB_Default_Qos to macro