nasa/cFE #1181: ES Globals need cleanup

jphickey, 2021-02-26:
**Is your feature request related to a problem? Please describe.**
The ES core app is quite disorganized in terms of its internal headers (what defines what) and global data structures.  In particular there are three global data objects in this module.

https://github.com/nasa/cFE/blob/c708d64bf2b25500cd0308a700aa4ee0b3fd5909/fsw/cfe-core/src/es/cfe_es_task.h#L148
https://github.com/nasa/cFE/blob/c708d64bf2b25500cd0308a700aa4ee0b3fd5909/fsw/cfe-core/src/es/cfe_es_global.h#L161
https://github.com/nasa/cFE/blob/c708d64bf2b25500cd0308a700aa4ee0b3fd5909/fsw/cfe-core/src/es/cfe_es_global.h#L166

**Describe the solution you'd like**
Make a single `CFE_ES_Global` object as was done for other apps in PR #1124.  Further clean up/re-org the internal headers to create a more logical/consistent content would be nice.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2021-02-26:
Just a guess - probably not all cleared at initialization either?  Good catch, adding to the Caelum target.

jphickey, 2021-02-26:
I can do this but since it involves files I'm already moving around in #972 -- I need to postpone/defer until after that one is done (i.e. one re-org at a time).

jphickey, 2021-02-26:
And yes, looks like `CFE_ES_Global` is actively cleared, the task data is not.

PR #1254 (2021-03-29): Fix #1181, global variable cleanup
**Describe the contribution**
Consolidate all ES global variables under a single CFE_ES_Global.

Removes the separate CFE_ES_TaskData as well as some random pointers
that were stored at global scope.

All references adjusted accordingly (search and replace).

Fixes #1181

**Testing performed**
Build and sanity check CFE, run all unit tests

**Expected behavior changes**
None, internal cleanup localized to ES implementation

**System(s) tested on**
Ubuntu 20.04

**Additional context**
The following ES variables were instantiated separately at global scope with [... cut]
commit 3397016d (2021-03-29): Merge pull request #1254 from jphickey/fix-1181-global-cleanup

Fix #1181, global variable cleanup
commit e8f73aa5 (2021-03-29): Fix #1181, global variable cleanup

Consolidate all ES global variables under a single CFE_ES_Global.

Removes the separate CFE_ES_TaskData as well as some random pointers
that were stored at global scope.

All references adjusted accordingly (search and replace).
PR #1258 (2021-04-01): cFE Integration candidate: 2021-03-30
**Describe the contribution**
Fix #1180, Removes impossible conditions
Fix #1002, Remove deprecated elements 
Fix #1252, Squash potentially uninitialized variable warnings
Fix #1185, Squash possible uninitialized variable false alarms
Fix #1246, Typo in CFE_TBL_Validate AppName
Fix #1192, Initialize TotalMsgSize in CFE_SB_GetUserDataLength 
Fix #1187, Increment CreatePipeErrorCounter for all create pipe errors
Fix #1186, Remove useless assignments/checks
Fix #1239, scrub include header guards 
Fix #1194, check for NULL in SlotUsed helpers 
Fix #1261, remov [... cut]
closed 2021-04-01 by astrogeco