nasa/cFE #1184: Infinite loop false alarm in unit test (easy fix) - static analysis warning

skliper, 2021-02-26:
**Is your feature request related to a problem? Please describe.**
Static analysis warning on `i` being int8.  Can be avoided if defined as size_t.
https://github.com/nasa/cFE/blob/c708d64bf2b25500cd0308a700aa4ee0b3fd5909/fsw/cfe-core/unit-test/ut_support.c#L82-L90

Based on logic strlen(UT_subsys) will never be bigger than int8 (unless someone redefines UT_subsys as something unreasonable) so this is a false alarm but easy to fix and better pattern to follow.

**Describe the solution you'd like**
Change to size_t

**Describe alternatives you've considered**
None

**Additional context**
Please wait until #972 is resolved.

**Requester Info**
Jacob Hageman - NASA/GSFC

himanshu007-creator, 2021-04-13:
I would like to work on this issue.

himanshu007-creator, 2021-04-28:
@skliper Hi, as #972 is closed, so i assume it is resolved. So should i go for a PR for this issue?

skliper, 2021-04-28:
@himanshu007-creator Yes please!  You'll need to follow the PR template instructions to fill out and deliver a CLA before the PR can be merged.  Thanks!

PR #1444 (2021-06-30): Fix #1184, Change index type to resolve infinite loop warning
**Describe the contribution**
A clear and concise description of what the contribution is.
-  Fixes #1184 

**Testing performed**
Steps taken to test the contribution:
1. Build steps '...'
1. Execution steps '...'

**Expected behavior changes**
A clear and concise description of how this contribution will change behavior and level of impact.
 - API Change: xxx (if applicable)
 - Behavior Change: xxx (if applicable)
 - Or no impact to behavior

**System(s) tested on**
 - Hardware: [e.g. PC, SP0, MCP750]
 - OS: [e.g. Ubuntu 18.04, RTE [... cut]
commit 6868369a (2021-06-30): Fix #1184, Change type to resolve loop warning

Static Analysis Warning - Infinite loop false alarm in unit test
commit e74b4fe3 (2021-06-30): Merge pull request #1444 from himanshu007-creator/patch-1

Fix #1184, Change index type to resolve infinite loop warning
PR #1654 (2021-07-07): cFE Integration candidate: 2021-06-29
## Context

Part of <https://github.com/nasa/cFS/pull/287>

## Tests 

cFE Checks <https://github.com/nasa/cFE/pull/1654/checks> 
Bundle Checks <https://github.com/nasa/cFS/pull/287/checks>

## Authors

@zanzaben 
@pepepr08 
@himanshu007-creator

## Description

### PR #1444
Fix #1184, Change index type to resolve infinite loop warning 

### PR #1653 
Fix #468, Adding coverage for cfe_es_task.c 

### PR #1638

Fix #1480, Expand FS Header Functional tests.
closed 2021-07-07 by astrogeco