nasa/cFE #1195: EVS_GenerateEventTelemetry doesn't handle vsnprintf error cases

skliper, 2021-03-01:
**Is your feature request related to a problem? Please describe.**
vsnprintf can return negative error values, but is compared to unsigned int to handle truncation:
https://github.com/nasa/cFE/blob/672b2dc4f76aa8535eb6b327b73f273c4cf87bb5/fsw/cfe-core/src/evs/cfe_evs_utils.c#L414-L417

I wouldn't call this a bug (will just pass the initialized to zero string), but might be worth a unique message?

**Describe the solution you'd like**
Explicitly handle failure (and cast for comparison)

**Describe alternatives you've considered**
Place termination character at the start? Any other way to provide clues.

**Additional context**
Static analysis warning for coercion alters value.

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2021-03-17:
Looking for opinions - could just cast sizeof to (int) which wouldn't do anything (message would still be empty, which is current behavior), could write something unique to message, could do an additional debug event (recursion), could write to the system log... simple "cFE_EVS_SendEvent vsnprintf error %d", or strerror(errno), or explain_vsnprintf().

jphickey, 2021-03-17:
It wouldn't hurt to first test for a negative result before the truncation test at line 417.  This way it will be ensured that the truncation test does not implicitly do a sign conversion.

As far as what to do -- it really shouldn't fail -- so anything is highly unlikely to occur.  Perhaps a WriteToSyslog, but that uses the same function (I think).  Just return early and skip the rest maybe?

PR #1235 (2021-03-18): Fix #1195, Avoid implicit conversion from vsnprintf errors
**Describe the contribution**
Fix #1195 - adds cast to avoid implicit conversion on a vsnprintf error, and comments to explain

**Testing performed**
Build/run unit tests (note #1224)

**Expected behavior changes**
No real change, vsnprintf should never error but does avoid static analysis warnings.

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: cFS Bundle main + this commit

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of  [... cut]
commit 10747de6 (2021-03-18): Fix #1195, Avoid implicit conversion from vsnprintf errors
commit 68cd7885 (2021-03-18): Merge pull request #1235 from skliper/fix1195-vsnprintf_error

Fix #1195, Avoid implicit conversion from vsnprintf errors
PR #1243 (2021-03-22): cFE Integration candidate: 2021-03-23
## Describe the contribution
Fix #1217, bit order macros
Fix #1033, removes false child tasks statement.
Fix #1195, adds cast to avoid implicit conversion on a vsnprintf error
Fix #590, Clarify the documentation on SB MsgId regarding requirements for command and telemetry messages
Fix #1208, Cast isspace input to unsigned char to avoid undefined behavior
Fix #1204, Updated message module, msgid v1 to use mask instead of cast to alter value (and added comment)
Fix #1200,  Coercion alters value caused by incorrect type - static analysis warning
Fix #1221, Us [... cut]
closed 2021-03-22 by astrogeco