nasa/cFE #1201: Question: Difference between WriteToSysLog or SendEvent

moafandi0, 2021-03-02:
Hi, I was reading through the application developers guide and I read "_Developers should make use of the Event Services CFE_EVS_SendEvent whenever possible. If, however, there is a significant Event that cannot be recorded using the CFE_EVS_SendEvent function, then the Developer can use the CFE_ES_WriteToSysLog function._ "

I was wondering what difference does it make if I use WriteToSysLog or SendEvent if my main goal is to print something to the terminal? Can I just always use WriteToSysLog?

Thanks!

astrogeco, 2021-03-02:
Hi! Welcome to our issue tracker. Can you post this question on our community mailing list? You can sign up at 
https://lists.nasa.gov/mailman/listinfo/cfs-community

moafandi0, 2021-03-03:
I just did. Thanks!

skliper, 2021-03-03:
CFE_EVS_SendEvent should be used by apps since they can be reported directly in TLM and be managed/filtered by type (event type) and uniquely identifiable (event id).  CFE_ES_WriteToSysLog is really for when EVS is not available, they are not transmitted directly in TLM (although you can dump and grab the log).  Occasionally libraries utilize CFE_ES_WriteToSysLog, but return codes should be the primary method for indicating errors (the calling App should report a unique event for failures based on the return code).  CFE_ES_WriteToSysLog is more free-form and really a last resort, and often conveys much less information (since any context information has to be added manually).

skliper, 2021-03-03:
Note both CFE_EVS_SendEvent and CFE_ES_WriteToSysLog are logged (separately), so CFE_EVS_SendEvent is really the way to go.

astrogeco, 2021-03-03:
Closing: conversation continued on mailing list

closed 2021-03-03 by astrogeco