nasa/cFE #1209: Implement single endianness handling pattern

skliper, 2021-03-04:
**Is your feature request related to a problem? Please describe.**
Multiple ways to handle target endian in various contexts:
fsw/cfe-core/src/es/cfe_es_perf.c:CFE_ES_SetupPerfVariables - determines it at runtime
fsw/cfe-core/src/tbl/cfe_tbl_internal.c:CFE_TBL_ReadHeaders - determines it at runtime
fsw/cfe-core/src/tbl/cfe_tbl_task_cmds.c:CFE_TBL_DumpToFile - determines it at runtime
fsw/cfe-core/src/fs/cfe_fs_api.c:CFE_FS_ReadHeader (and many other functions in here) - determines it at runtime

fsw/cfe-core/src/inc/ccsds.h has macros for conversion (CFE_MAKE_BIG*) who’s implementation depends on SOFTWARE_BIG_BIT_ORDER
There’s an endian flag in the CCSDS header
osal/src/os/inc/common_types.h defines either SOFTWARE_LITTLE_BIT_ORDER or SOFTWARE_BIG_BIT_ORDER based on 8 possible defines

cfe/cmake/sample_defs/cpu1_platform_cfg.h defines CFE_PLATFORM_ENDIAN as either CCSDS_LITTLE_ENDIAN or CCSDS_BIG_ENDIAN, and also has a separately configurable CFE_PLATFORM_TIME_CFG_BIGENDIAN

CCSDS extended header has an endian bit

Then there’s all the different ways the various defines are used.

**Describe the solution you'd like**
See https://github.com/nasa/cFE/issues/202#issuecomment-536680998 for a compile time suggestion:
#define IS_LITTLE_ENDIAN (((union {unsigned x; char c;}){1}).c)

**Describe alternatives you've considered**
None

**Additional context**
Triggered from email discussion on setting another endian flag in a toolchain file for an app to use

**Requester Info**
Jacob Hageman - NASA/GSFC

Ping - @klystron78 @ejtimmon @wmoleski

wmoleski, 2021-03-04:
Jake,

                This is defined in my sample_defs/cpu1_platform_cfg.h file. However, CF seems to be looking for an additional define.

                Also, I thought the MCP750 was BIG ENDIAN with vXWorks. The above file has LITTLE set in my repo.

Walt

Walt Moleski                  Phone: 301-286-7633
NASA/GSFC Code 582   Fax: 301-286-1768
Greenbelt, MD 20771
email: Walter.F.Moleski@nasa.gov<mailto:Walter.F.Moleski@nasa.gov>

See #202 (comment)<https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fnasa%2FcFE%2Fissues%2F202%23issuecomment-536680998&data=04%7C01%7Cwalter.f.moleski%40nasa.gov%7Cc4f7b4ceb01d4bb7fd6b08d8df3d4565%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C637504801545877402%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=81rVPefvdLbnWukGS1h01UhMe6noOnKC9jMkICEE5oo%3D&reserved=0> for a compile time suggestion:

jphickey, 2021-03-04:
The challenge here is that the C standard (and library) don't want to expose endianness.  They actively discourage one from writing endian-specific code, and there is no standardized "marker" to say what endianness the current CPU has.  In particular a machine is allowed to arrange the bits however it pleases and still be compliant (see PDP-11).  AFAIK There is even a variant that takes the weird PDP-11 order and reverses it.  While none of these are common, they do exist.  

The problem with any "endian" check is that it is usually binary on the common ones (big or little), and does not account for other variations.

Rather than introducing another endianness-check to rule them all, I'd rather see migration away from any sort of endian-specific code/dependencies.

FWIW, the `SOFTWARE_LITTLE_BIT_ORDER` (a misnomer because its actually byte order) was intended for compile-time checks where the union trick does not work.