nasa/cFE #1215: Clean up ES task/app registration

jphickey, 2021-03-10:
**Is your feature request related to a problem? Please describe.**
In nasa/osal#853 it is proposed that `OS_TaskRegister()` be finally deprecated/removed.

But this function is invoked by two places in CFE ES that follow a similar pattern: `CFE_ES_RegisterApp()` and `CFE_ES_RegisterChildTask()`.

**Describe the solution you'd like**
At a minimum, calls to `OS_TaskRegister()` must be removed to allow the function to be deprecated.

Furthermore there is already a task startup wrapper in ES that can be used to call/handle setting up environment (`CFE_PSP_SetDefaultExceptionEnvironment()`) meaning that these two functions themselves can also be deprecated - basically using the same design pattern as OSAL uses such that we don't need to burden apps with calling this extra function.  This design is simpler and less error prone.

**Additional context**
See nasa/osal#853

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-03-10:
Ping @skliper , @ejtimmon , @acudmore , @jwilmot ....

This is the only ref to OS_TaskRegister() that I found but interestingly we have basically the same situation in CFE ES - a couple functions that have effectively become no-ops - or at least unnecessary - but remain in the API for compatibility reasons.

However unlike OS_TaskRegister it looks like these ARE invoked by many CFS apps - so there are definite backward compatibility issues to removing/deprecating it.

This makes me think we should just leave it alone - is it that bad to have a couple no-op functions laying around for backward compatibility?

skliper, 2021-03-10:
I find it more confusing to have an API that's half utilized, possibly partly documented, doesn't actually do anything, not really clear what the plan is for the future (will we leave it forever?), etc.  I'd rather a quick/hard break than the uncertainty the noops cause.  Unless we can fully utilize/document and very clearly state to keep these calls in (for future development or whatever), as it it's a bug if they are missing.

Basically half way is worse (in my view) than hard yes or hard no.

skliper, 2021-03-17:
CCB - if we are going to do it, do it now.  No hard objects, lets remove.

skliper, 2021-03-18:
Also noted on community-email list the comment below is misleading.  As long as we are removing then no additional action needed.  If for some reason this stays, should be updated.

https://github.com/nasa/cFE/blob/fa10af72f95a3a36b65352f0688498449487265f/modules/es/fsw/src/cfe_es_api.c#L663

ping @johnphamngc

commit 66e24636 (2021-03-29): Merge pull request #1250 from jphickey/fix-1215-task-registration

Fix #1215, remove task registration calls #1250
commit 8c70e950 (2021-03-29): Fix #1215, remove task registration calls

Explicit task registration is no longer necessary, since all
required actions can be done before invoking the entry point.

This moves the invocation of CFE_PSP_AttachExceptions() from
the registration function to the pre-entry function, this was
the only remaining action in task registration.

All references to task registration in code, docs, and tests
are removed.
PR #1258 (2021-04-01): cFE Integration candidate: 2021-03-30
**Describe the contribution**
Fix #1180, Removes impossible conditions
Fix #1002, Remove deprecated elements 
Fix #1252, Squash potentially uninitialized variable warnings
Fix #1185, Squash possible uninitialized variable false alarms
Fix #1246, Typo in CFE_TBL_Validate AppName
Fix #1192, Initialize TotalMsgSize in CFE_SB_GetUserDataLength 
Fix #1187, Increment CreatePipeErrorCounter for all create pipe errors
Fix #1186, Remove useless assignments/checks
Fix #1239, scrub include header guards 
Fix #1194, check for NULL in SlotUsed helpers 
Fix #1261, remov [... cut]
closed 2021-04-01 by astrogeco