nasa/cFE #1216: Update directory diagrams in application developer guide

jphickey, 2021-03-10:
**Is your feature request related to a problem? Please describe.**
After PR #1203 the directory structure diagrams in the user guide are in need of an update.  Only the text was updated in that PR.

**Describe the solution you'd like**
Diagrams should be updated to match the text.

**Additional context**
See #1203

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2021-03-10:
Note I'd be fine with dropping pictures and using a simpler layout.

jphickey, 2021-03-10:
The other option is to just copy/paste the output of a `tree -d` command (which is what I put into previous comments).

Fast and easy to update when needed, way easier than graphics....

skliper, 2021-03-10:
> `tree -d`

Perfect.

PR #1903 (2021-09-07): Fix #1216, Update Directory Diagrams in App Developer Guide
**Describe the contribution**
Fixes #1216 

**Expected behavior changes**
Used tree to provide a tree view of cFS, cFE, Build, and Apps. 

**Additional context**
Did not add CS directory. 

**Contributor Info - All information REQUIRED for consideration of pull request**
Ariel Adams, ASRC Federal
commit 50c139fb (2021-09-07): Fix #1216, Update Directory Diagrams in App Developer Guide
commit fdea0a27 (2021-09-07): Merge pull request #1903 from ArielSAdamsNASA/fix-1216-update-directory-diagrams-app-developer-guide

Fix #1216, Update Directory Diagrams in App Developer Guide
PR #1939 (2021-09-10): cFE Integration candidate: 2021-09-07
**Describe the contribution**

PR #1903

-  Fix #1216, Update Directory Diagrams in App Developer Guide

PR #1922 

- Fix #1905, Improve SB coverage 

PR #1936 

- Improve ES code coverage 
- Fix #1878 
- Fix #1696 

PR #1938 

- Fix #1906, Enforce minimum code coverage 

PR #1940 

Update dev guide for MSG and SBR

- Fix #800 
- Fix #961

PR #1941 

- Fix #1931 Support unimplemented features in MSG ft 

**Testing performed**
cFE Checks <https://github.com/nasa/cFE/pull/1939/checks>
cFS Bundle Checks <https://github.com/nasa/cFS/pull/351/checks>

**Expecte [... cut]
closed 2021-09-10 by astrogeco