nasa/cFE #1223: Entry Point function name too long

zanzaben, 2021-03-16:
**Describe the bug**
When trying to run the cfe test runner I got this error 

1980-012-14:03:20.46280 ES Startup: Loading file: /cf/cfe_testrunner.so, APP: TESTRUN_APP
OS_GenericSymbolLookup_Impl():118:Error: CFE_TestRunner_AppM: ./cf/cfe_testrunner.so: undefined symbol: CFE_TestRunner_AppM
1980-012-14:03:20.46306 ES Startup: Could not find symbol:CFE_TestRunner_AppM. EC = 0xFFFFFFFF

It should be trying to use CFE_TestRunner_AppMain but is getting truncated at 19 characters. 
When I shortened the function name it worked fine. 

**To Reproduce**
1. add the start test scripts for the startup test. (found here [https://github.com/nasa/cFE/blob/2b8844871d374219cef7c0026e6c43460e11b22b/docs/README_functionaltest.md])
2. Start it. 

**Expected behavior**
It should run the tests. 

**System observed on:**
Ubuntu 20.04

**Reporter Info**
Alex Campbell GSFC

zanzaben, 2021-03-16:
@jphickey should CFE_TestRunner_AppMain be shortened to something like CFE_TestRunner_AM or should it be able to work with function names longer than 19 characters?

jphickey, 2021-03-17:
Since AppMain suffix is used elsewhere (sample apps, LAB apps, etc), how about `CFETR_AppMain`?

skliper, 2021-03-17:
I like CFE_TR_AppMain since it's provided by CFE

zanzaben, 2021-03-17:
I think CFE_TR_AppMain is the most consistent with other function names.

PR #1242 (2021-03-29): Fix #1223, #1264 shorten TestRunner function name and update cfe_es_startup.scr docs
**Describe the contribution**
Fixes #1223
Change the name of the testrunner main function to be less than 20 characters. 

Fixes #1264
Use simple filenames. 

**Testing performed**
Build and run unit test

**Expected behavior changes**
No impact to behavior

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campbell GSFC
commit 096127ba (2021-03-29): Fix #1223, shorten TestRunner function name
commit b46d1642 (2021-03-29): Merge pull request #1242 from zanzaben/fix1223_Testrunner_name_too_long

Fix #1223, #1264 shorten TestRunner function name and update cfe_es_startup.scr docs
PR #1258 (2021-04-01): cFE Integration candidate: 2021-03-30
**Describe the contribution**
Fix #1180, Removes impossible conditions
Fix #1002, Remove deprecated elements 
Fix #1252, Squash potentially uninitialized variable warnings
Fix #1185, Squash possible uninitialized variable false alarms
Fix #1246, Typo in CFE_TBL_Validate AppName
Fix #1192, Initialize TotalMsgSize in CFE_SB_GetUserDataLength 
Fix #1187, Increment CreatePipeErrorCounter for all create pipe errors
Fix #1186, Remove useless assignments/checks
Fix #1239, scrub include header guards 
Fix #1194, check for NULL in SlotUsed helpers 
Fix #1261, remov [... cut]
closed 2021-04-01 by astrogeco