nasa/cFE #1249: CFE_ES_CopyModuleAddressInfo not working

zanzaben, 2021-03-23:
**Describe the bug**
When getting the app info, the AddressesAreValid is false. That field is set by CopyModuleAddressInfo when it's Return code is not OS_SUCCESS. The return code it -16 Invalid ID

**To Reproduce**
Steps to reproduce the behavior:
Call CFE_ES_GetAppInfo 
See that AddressesAreValid are set to false. 

**Expected behavior**
It shouldn't be setting them to false. 

**System observed on:**
Ubuntu 20.04

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Alex Campbell GSFC

jphickey, 2021-03-23:
Note that not all operating systems allow OSAL to get address info of modules.  VxWorks allows it but POSIX does not have a standard interface to get this info.  This is why the OSAL function `OS_ModuleInfo()` returns success - all the _other_ info in the returned structure is valid, but the address info is not - because there is no way to get the info.

This sounds like correct behavior when running on Ubuntu/Linux.  When you run on VxWorks the AddressesAreValid flag should be true.

skliper, 2021-03-23:
Sounds like the test should do a NA if the AddressesAreValid == false and skip non-zero checks, otherwise do the non-zero checks.

zanzaben, 2021-03-25:
After discussing with @jphickey this is all working as intended.

closed 2021-03-25 by zanzaben