nasa/cFE #1266: CFE_Assert should use SendEvent

zanzaben, 2021-03-25:
**Is your feature request related to a problem? Please describe.**
CFE_Assert currently has all it's outputs done through CFE_ES_WriteToSysLog which makes them hard to read and failures can get lost in all the noise. 

**Describe the solution you'd like**
CFE_Assert could instead send events through the test runner, which would give a lot more flexibility to the output. 

**Describe alternatives you've considered**
We just leave it as it is. 

**Additional context**
https://github.com/nasa/cFE/issues/94 could then be a further enhancement of this done at a later time. 

**Requester Info**
Alex Campbell GSFC

jphickey, 2021-03-25:
Also worth noting that the syslog is usually of limited size, so if the test cycle has lots of tests, it is easy for the log to overflow and early messages will get completely lost.

I've evaluated this idea and it is not totally trivial because the events have to be sent by the cfe_testrunner app, not cfe_assert itself (because libraries cannot use event services directly) but that shouldn't be too hard to overcome, a simple notifier callback paradigm should get the job done.

skliper, 2021-03-25:
Could "cheat" in this non-flight code context and just use a single ID, either passed in (maybe w/ macro) or initialized.

jphickey, 2021-03-25:
I don't see any need to cheat, its easy enough to let cfe_assert call back into test runner.

The other issue to consider is startup order - cfe_assert allows conditions to be asserted during startup phase, and these might occur before event services is even ready to send events.  These will always have to go to syslog.  

Using a callback also solves the problem by allowing these early asserts to go to syslog but then switching to the events for the bulk/normal tests.  Could also send early failures as events in a deferred fashion.  Lots of options.

jphickey, 2021-04-01:
I'll submit something as a proof of concept here

PR #1276 (2021-04-12): Fix #1266, use events for CFE test asserts
**Describe the contribution**
Reports test failures as CFE events, which allows for more processing capability, and allows failures to be recieved externally (e.g. ground system).

Fixes #1266

**Testing performed**
Update startup script to load the CFE functional test libs + apps and execute CFE to run the tests.  

**Expected behavior changes**
Test status messages sent as Events rather than Syslog.

**System(s) tested on**
Ubuntu 20.04

**Additional context**
Also tested in conjunction with #1251.

**Contributor Info - All information REQUIRED for  [... cut]
commit 4a20389d (2021-04-12): Fix #1266, use events for CFE test asserts

Reports test failures as CFE events, which allows for more processing
capability, and allows failures to be recieved externally (e.g. ground system).
commit f8102e9b (2021-04-12): Merge pull request #1276 from jphickey/fix-1266-assert-events

Fix #1266, use events for CFE test asserts
PR #1295 (2021-04-12): cFE Integration Candidate: 2021-04-13
## Description

### PR #1276 

Fix #1266, use events for CFE test asserts 

Reports test failures as CFE events. Test status messages are now sent as Events rather than Syslog. This allows for more processing capability, and allows failures to be received externally (e.g. ground system). 

## Context

Part of <https://github.com/nasa/cFS/pull/242>

## Tests

cFE checks <https://github.com/nasa/cFE/pull/1295/checks>
cFS bundle checks <https://github.com/nasa/cFS/pull/242/checks>

## Author

@jphickey
closed 2021-04-12 by astrogeco