nasa/cFE #1277: Duplicate Function in Readme

ArielSAdamsNASA, 2021-04-01:
**Describe the bug**
In the version history of the readme, it states: 
> Removes the now-unnecessary CFE_SB_ZeroCopyHandle_t type and all APIs that refer or require it .Replaces CFE_SB_ZeroCopyGetPtr() and CFE_SB_ZeroCopyGetPtr() with two new simplified functions CFE_SB_AllocateMessageBuffer() and CFE_SB_ReleaseMessageBuffer() , respectively. These new functions do not use a separate handle. Updates the CFE_SB_TransmitBuffer() API to also remove the handle. Does affect public APIs. 

The issue is that the version history is stating that the same function, CFE_SB_ZeroCopyGetPtr, is being replaced by two different functions. 

In reality, CFE_SB_ZeroCopyGetPtr was replaced by CFE_SB_AllocateMessageBuffer. CFE_SB_ZeroCopyReleasePtr and CFE_SB_ZeroCopySend, not CFE_SB_ZeroCopyGetPtr, was replaced by CFE_SB_ReleaseMessageBuffer.  

**Expected behavior**
State that CFE_SB_ZeroCopyReleasePtr and CFE_SB_ZeroCopySend, not CFE_SB_ZeroCopyGetPtr, was replaced by CFE_SB_ReleaseMessageBuffer. 

**Code snips**
https://github.com/nasa/cFE/pull/1257/files 

**Reporter Info**
Ariel Adams, ASRC Federal

skliper, 2021-04-02:
Note `CFE_SB_ZeroCopySend` was replaced by `CFE_SB_TransmitBuffer`.  `CFE_SB_ZeroCopyReleasePtr` was replaced by `CFE_SB_ReleaseMessageBuffer`.

astrogeco, 2021-04-02:
Good catch, I think this was a copy paste error from a typo in the original PR.

astrogeco, 2021-04-02:
Just fixed it in the comments for #1257 and #1258. Created a new PR, https://github.com/nasa/cFE/pull/1278, to fix it.

PR #1278 (2021-04-02): Fix #1277, ZeroCopy typo in ReadMe
**Describe the contribution**
Fix #1277, ZeroCopy typo in ReadMe

**Testing performed**
Docs only

**Expected behavior changes**
None

**Additional context**
None

**Third party code**
Nonoe

**Contributor Info - All information REQUIRED for consideration of pull request**
@astrogeco
commit acaf1356 (2021-04-02): Merge pull request #1278 from nasa/fix-1277-readme-zerocopy-typo

Fix #1277, ZeroCopy typo in ReadMe
commit e294941b (2021-04-02): Fix #1277, ZeroCopy typo in ReadMe

Fixes mislabeled function in v6.8.0-rc1+dev484 change description
PR #1279 (2021-04-02): cFE Integration candidate: 2021-04-02
## Change Description and Impacts

### PR #1274 

Fix #1273, remove PspConfig global object 

Removes `PspConfig` member from the `GLOBAL_CONFIGDATA` object. Updates the only remaining reference to this object inside the `CFE_PSP_VERSION` and uses the new Version API instead. Updates the OSAL version print to use the version API call and CFE uses the macro directly.

### PR #1278 

Fix #1277, ZeroCopy typo in ReadMe.md

Replaces duplicate mention of the removed `CFE_SB_ZeroCopyGetPtr` with the correct removal target of `CFE_SB_ZeroCopyReleasePtr`

## Tests [... cut]
closed 2021-04-02 by astrogeco