nasa/cFE #1287: Module list is duplicated/hard coded so can't be modified

skliper, 2021-04-08:
**Is your feature request related to a problem? Please describe.**
Part of the intent of modules was to be able to replace them.  Hard-coding of the module list at:
https://github.com/nasa/cFE/blob/eaa343c2e5d457f7d04b65ada8881b24e4fb46bc/modules/core_api/CMakeLists.txt#L18

defeats this capability.

**Describe the solution you'd like**
Define the module list in a way it can be modified/customized by configuration.

**Describe alternatives you've considered**
None

**Additional context**
Currently breaks message header customization.

**Requester Info**
Jacob Hageman - NASA/GSFC

Ping - @excaliburtb

jphickey, 2021-04-08:
That's not what this is - this is just for the API.  One can still customize the implementation of any module.

The API needs to have some standards/baselines - while the implementation can be overridden, it needs to conform to the same API.  The list in this file is what defines the notion of "cfe core API" as a whole.

I can see where this might not work if the name was changed, though.  Is the intent that one not only wants to override the module implementation, but also change its name i.e. to have the "tbl" functions provided by a module not called "tbl"?

skliper, 2021-04-08:
It's pulling in the include directories and compile definitions of the hard coded modules... that's broken.

jphickey, 2021-04-08:
Not necessarily the ones in `modules/` though - if there is a user-provided implementation, that will be used instead, and it should be the interface definitions from the user-provided implementation, not the default one.

The assumption here is that it (i.e. the user provided module) will have the same target name, is that not correct?

skliper, 2021-04-08:
Correct, projects should be able to use whatever names they want.  The previously working example (see https://github.com/nasa/cFE/issues/879).

jphickey, 2021-04-08:
They can remove/add/replace modules, that all works.  The problem is that we need to know whether that module is intended to provide something that is conceptually part of the "core_api" or if it is some user/mission thing that is not part of the core api.

Case in port, for #879 - we don't know the intent of a module that is named `custom_msg`  - that is, whether it is intended to provide the `msg` interface that is part of `core_api` or whether it is just some other app/lib that happens to be named `custom_msg` and does something else.

This (perhaps incorrectly) assumed that if one was overriding the `msg` lib, it would still be called `msg`, which is the only way to know if it is intended to fulfill that role.

Would an alias target be OK?  That might work...

jphickey, 2021-04-08:
To clarify that "alias" thought a bit - what I'd suggest is that in the CMakeLists.txt for the `custom_msg` module, it does this:

    # Indicate that this is intended to fulfill the role of "msg" core interface
    add_library(custom_msg ALIAS msg)

PR #1290 (2021-04-22): Fix #1287, split interface and implementation modules
**Describe the contribution**
Separate the list of CFE core interface modules (e.g. core_api) from the list of CFE core implementation modules (e.g. msg).  This allows the content of core_api to be expanded to include any additional modules the user has added to cFE core locally.

Fixes #1287 

**Testing performed**
Build and sanity check CFE
Run all unit tests

**Expected behavior changes**
No FSW change, only build system.
The "core_api" interface module will now be expanded to include anything that is in the `MISSION_CORE_MODULES` list,  [... cut]
commit 1a72fad3 (2021-04-22): Fix #1287, split interface and implementation modules

Separate the list of CFE core interface modules (e.g. core_api) from
the list of CFE core implementation modules (e.g. msg).  This allows
the content of core_api to be expanded to include any additional
modules the user has added to cFE core locally.
commit 9178dfe0 (2021-04-22): Merge pull request #1290 from jphickey/fix-1287-core_api

Fix #1287, split interface and implementation modules
PR #1406 (2021-04-22): cFE Integration candidate: 2021-04-20
## Description

### PR #1290

 Fix #1287, split interface and implementation modules

Separates the list of CFE core interface modules (e.g. core_api) from the list of CFE core implementation modules (e.g. msg). This allows the content of core_api to be expanded to locally include any additional modules the user has added to cFE core via the `MISSION_CORE_MODULES` list.

### PR #1376 

Fix #1337, add docs to `CFE_ES_RegisterCDS()` regarding clearing 

Note in the documentation for this function that it is the responsibility of the calling app to clear/fill [... cut]
closed 2021-04-22 by astrogeco