nasa/cFE #1293: CFE_PSP_GetTime conversion to MET

mogar, 2021-04-09:
**Is your feature request related to a problem? Please describe.**

We have a hardware clock in our system that we plan to use as the source clock for `CFE_PSP_GetTime`.

We also have an FPGA system that latches the hardware clock into a register on certain events. We'd like to read the register value after it's latched, and then convert it to a cFS MET. Does CFE Time services already have a routine for converting PSP Times to METs?

**Describe the solution you'd like**

A function that I can pass an old LatchClock time value into and get back a valid `CFE_TIME_Reference_t` value. Something similar to `CFE_TIME_GetReference(CFE_TIME_Reference_t *Reference)`, but with an additional input that replaces `Reference->CurrentLatch`.

**Describe alternatives you've considered**

Our primary alternative at this point is to add code to the PSP that updates our hardware clock via TATT commands. Any other ideas would also be welcome.

**Requester Info**

Morgan Redfield
Morgan.Redfield@astrobotic.com
Astrobotic

jphickey, 2021-04-09:
IIRC the CFE_TIME_GetReference() function just samples the PSP time, then offsets this to the `AtToneMET` value from the last time tone message.

If you read the current reference via CFE_TIME_GetReference(), the output struct should have all the relevant values - CurrentLatch (the time from PSP), CurrentMET, and AtToneMET.  You can simply take the difference between CurrentLatch and your captured latch time and add/subtract this to the CurrentMET to get the correlating MET value.

But -- importantly - this requires that PSP hardware clock is monotonic / always increasing and does not wrap back to 0.  You could still do this (theoretically at least) with a HW clock that wraps around, but you have to consider this with every add/subtract and are limited to 1/2 of the wrap period or things get ambiguous.

mogar, 2021-04-13:
That makes sense. Thank you.

closed 2021-04-13 by mogar