nasa/cFE #1307: Cleanup relative to additional compiler warning flags

skliper, 2021-04-13:
**Is your feature request related to a problem? Please describe.**
Some projects prefer additional flags that trigger warnings in cFE.

Current set:
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/cmake/sample_defs/arch_build_custom.cmake#L29-L36

Possible additions for consideration:
-Wextra -Wmissing-prototypes -Wimplicit-function-declaration -Wnested-externs

**Describe the solution you'd like**
Analyze, fix what makes sense.

**Describe alternatives you've considered**
Not required to add to default list, but should compile cleanly if enabled (for those that make sense)

**Additional context**
From CFS-40 code review

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-04-13:
The `-Wextra` flag contains options that are likely to generate some nuisance warnings and also may vary between versions of gcc.  I swear in the past `-Wextra` enabled things like `-Wconversion` which we already decided against, yet the manpage doesn't list that one, so I guess its not.

At any rate it might be preferable to enable only specific warning types to give us more fine-grained control.  

I would suggest these as good candidates, at least:

```
-Wtype-limits
-Wignored-qualifiers
-Wuninitialized
-Wunused-but-set-parameter
-Wsign-compare
```

jphickey, 2021-04-13:
Also worth noting - although CFE itself doesn't do a lot of floating point, at one point I did work on software for a small processor that only had a single-precision hardware floating point unit, and the `double` type had to be emulated in software.  The `-Wdouble-promotion` warning is very valuable on this type of system.  

It might be worthwhile to ensure that if there are any uses of floating point arithmetic in the framework, that we aren't doing implicit double promotions unnecessarily.