nasa/cFE #1319: Remove `extern` from function prototypes

skliper, 2021-04-13:
**Is your feature request related to a problem? Please describe.**
Inconsistently applying extern to function prototypes

**Describe the solution you'd like**
Remove, it makes no difference.

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

himanshu007-creator, 2021-04-13:
I would like to work on this issue. Please guide me further

pavll, 2021-08-28:
@skliper 

I have removed all `extern` from all function prototypes in the project. Could this cause unforeseen problems (in conjunction with the cFS)?

skliper, 2021-08-30:
Not for the function prototypes, declaring as `extern` has no impact since it's the default. https://stackoverflow.com/questions/27447285/extern-on-function-prototypes.  Thanks for the contribution!

pavll, 2021-08-30:
Ok, great to know.

commit d81ad3d1 (2021-12-06): Fix #1319, Remove extern from function prototypes
closed 2021-12-06 by astrogeco