nasa/cFE #1324: Add null pointer check to CFE_ES_ReloadApp

skliper, 2021-04-13:
**Is your feature request related to a problem? Please describe.**
AppFIleName NULL not checked (or tested)
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/modules/es/fsw/src/cfe_es_api.c#L220-L226

**Describe the solution you'd like**
Add a NULL test, add the check (if needed)

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

zanzaben, 2021-04-14:
AppFileName is tested for null inside OS_stat which is the first place to use it.

skliper, 2021-05-20:
Switched to unit-test only, since just the functional check is needed for the NULL pointer.

zanzaben, 2021-07-15:
This will be covered by #806

PR #1722 (2021-08-06): Fix #806, Add application control functional tests
**Describe the contribution**
A clear and concise description of what the contribution is.
- Fixes #806
- Fixes #1324
- Adds functional tests for
  - CFE_ES_RestartApp
  - CFE_ES_ReloadApp
  - CFE_ES_DeleteApp

**Testing performed**
Steps taken to test the contribution:
1. Build CFE and run all functional tests

**Expected behavior changes**
A clear and concise description of how this contribution will change behavior and level of impact.
 - no impact to behavior

**System(s) tested on**
 - Hardware: PC
 - OS: Ubuntu 20.04

**Contributor Info  [... cut]
PR #1772 (2021-08-11): cFE Integration candidate: 2021-08-10
**Describe the contribution**

PR #1737
- Fix #1733, Move global count into test global struct. 

PR #1722 
- Fix #806, Add ES application control API functional tests 
- Fix #1324, Add null pointer check to CFE_ES_ReloadApp functional test 

PR #1753 
- Fix #813, Add Generic Counter API test 

PR #1744 

- Fix #1743, update coverage test to use UtAssert macros 

PR #1734
  - Fix #1685
  - Fix #1688 
  - Fix #1686 
  - Fix #1687 

PR #1716 

- Fix #1645, Add resource ID API functional tests 

PR #1764 

- Fix #1763, last char truncated in coverage log outp [... cut]
closed 2021-08-11 by astrogeco