nasa/cFE #1334: Add missing prototypes

skliper, 2021-04-13:
**Is your feature request related to a problem? Please describe.**
Enable -Wmissing-prototypes and observe warnings

Example identified in code review:
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/modules/es/fsw/src/cfe_es_apps.c#L1589

**Describe the solution you'd like**
Add prototype

**Describe alternatives you've considered**
None

**Additional context**
Code reivew, specific case related to #1307 

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2021-05-26:
Note there's 51 missing prototypes flagged by gcc in the main source code, 742 with unit tests enabled.  Vast majority are really static (but not declared as such).  Removing 7.0.0 milestone (future work).

thnkslprpt, 2022-10-26:
What is the preferred solution here?
1) Do nothing (i.e. if prototypes are not needed, nothing needs changing here)
2) Make functions not required outside file scope `static`
3) Add missing prototypes to header file  (let's say only for code for now, not test functions)

skliper, 2022-10-26:
I like option 1 for it's simplicity, elegance, and economy.  I'm not aware of any benefits to adding an unnecessary prototype would be.  Could close (up to @dmknutsen @dzbaker).

skliper, 2022-10-26:
I guess adding `static` would make it clear to the reader/reviewer, so no harm in that option.

dzbaker, 2022-10-26:
> I like option 1 for it's simplicity, elegance, and economy. I'm not aware of any benefits to adding an unnecessary prototype would be. Could close (up to @dmknutsen @dzbaker).

I like option 1 as well, although I don't feel strongly about it.

thnkslprpt, 2022-11-06:
It certainly doesn't get any simpler, elegant or economical than doing nothing...
I guess unless @dmknutsen disagrees, this can be closed with no changes required.

dmknutsen, 2022-11-07:
Option 1 sounds great!