nasa/cFE #1340: CFE_ES_DeleteCDS: zero out block when freed

skliper, 2021-04-13:
**Is your feature request related to a problem? Please describe.**
Block not cleared when freed:
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/modules/es/fsw/src/cfe_es_cds.c#L883

**Describe the solution you'd like**
Analyze, clear block if needed or document for future reference why it's not needed

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-04-19:
Similar to #1337, there are good reasons to avoid excessive writing to CDS, and the code has historically not done this.  My recommendation would be to not forcibly clear it.

However I did submit #1377 as a suggestion to make this check stronger, possibly an enhancement for future version

jphickey, 2021-04-19:
Honestly I'm not even sure how to document this.  Code should never use a block after deleting it, and there is never an assumption (or should never be an assumption) that memory will be cleared as part of a delete/free operation.  It's a case where actually saying this might be more confusing than saying nothing at all.... because by definition the memory block is _deleted_ ... it doesn't logically exist anymore, how can we document what the contents of something that doesn't exist are?

skliper, 2021-04-19:
Maybe just detail the fact the CDS entry gets deleted from the registry without modification to the actual memory block in the CDS  (or whatever language is appropriate).  This is helpful info for users that need to track write cycles.  I don't see it as a case of code using the block after delete, it's detailing what the "delete" does which can be important to managing the hardware.

jphickey, 2021-04-19:
Yeah, that can be done.  I was thinking along the lines of "document what is in the memory buffer after a call to `free()`" which just didn't make sense.  But this is more akin to deleting a file on disk, and to note that the underlying storage (physical sectors) are not actually wiped of the data.

PR #1384 (2021-04-28): Fix #1340, update documentation for CFE_ES_DeleteCDS
**Describe the contribution**
Noted that this does not actually wipe or erase the block, it only returns resources to the pool for re-use.

Fixes #1340

**Testing performed**
Build and sanity check CFE

**Expected behavior changes**
None, documentation change only.

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Joseph Hickey, Vantage Systems, Inc.
commit 1b91d335 (2021-04-28): Merge pull request #1384 from jphickey/fix-1340-delete-cds-doc

Fix #1340, update documentation for CFE_ES_DeleteCDS
commit 9ed820fb (2021-04-28): Fix #1340, update documentation for CFE_ES_DeleteCDS

Noted that this does not actually wipe or erase the block, it only
returns resources to the pool for re-use.
closed 2021-04-29 by astrogeco